espressif/ESP8266_RTOS_SDK #167: ESP-Mesh compatibility

albydnc, 2018-04-30:
Hi, I have seen on the website that esp8266 is compatible with esp-mesh, but it seems that all the references and docs are outdated, all the examples urls give 404.
Is it still a feature for this SoC? or is it abandoned?
I have been using some 3rd party mesh libraries, but the ESP32 mesh seems more advanced, with a lot of features that could come handy to my prototypes.

27-301, 2018-05-02:
hi, could you point which urls in espressif's website?

albydnc, 2018-05-02:
https://www.espressif.com/en/products/software/esp-mesh/overview
In related products appears esp8266ex

27-301, 2018-05-02:
En..., it maybe a mistake. Let them fix it.

albydnc, 2018-05-02:
https://www.espressif.com/sites/default/files/30a-esp8266_mesh_user_guide_en.pdf
There is also the guide (3rd result in Google search)

FayeY, 2018-05-22:
Hi, yes, it is the ESP32 mesh that be recommended to use. We suggest using ESP32 mesh instead, it is one of the major projects we are working on now. Thanks for your interest in ESP!

closed 2018-05-22 by FayeY