espressif/ESP8266_RTOS_SDK #170: ESP8266 SMD CHIP

inventvictor, 2018-05-02:
Hi 

Please I want to know if the ESP8266 SMD CHIP (the regular smd chip that sells in the market) comes preloaded with any firmware. If No, How is it possible to load it with a firmware? is it the normal way of flashing?

Thank you.

Look forward to hear from you soon.

Cheers.

27-301, 2018-05-02:
hi, for esp8266, there MUST need a spi flash, and the firmware will be stored in this flash, but not in esp8266 chip.

For modules, such as [esp-wroom02](http://espressif.com/sites/default/files/documentation/0c-esp-wroom-02_datasheet_en.pdf), you can use [esptool](https://github.com/espressif/esptool) or espressif's [Flash Download Tools](https://www.espressif.com/sites/default/files/tools/flash_download_tools_v3.6.4_0.rar).

If you design your own module or board, you can flash the spi flash firstly, then solder to board.

inventvictor, 2018-05-02:
Thanks for this @27-301 

So basically I can also solder the SPI flash to the board and then use esptool to program using a usb-uart converter right?

Please reply soon.

Thanks

closed 2018-05-24 by FayeY