espressif/ESP8266_RTOS_SDK #173: Instable WiFi

berkutta, 2018-05-05:
I'm facing the issue that the ESP8266 basically as sudden isn't able anymore to send TCP stuff anymore. I don't get a wifi event or any other stuff out of the debug UART. Also the wifi status doesn't change. After a Reset of the ESP itself everything is fine again for a certain time when the same issue occurs again.

UART Log:
```
Temperature: 24.625000
Loop: 4170
Wifi status: 5
... allocated socket
... connected
... socket send success

Temperature: 24.562500
Loop: 4171
Wifi status: 5
... allocated socket
... connected
... socket send success

Temperature: 24.562500
Loop: 4172
Wifi status: 5
... allocated socket
... socket connect failed errno=113
... socket send failed

Temperature: 24.562500
Loop: 4173
Wifi status: 5
... allocated socket
... socket connect failed errno=113
... socket send failed
```

27-301, 2018-05-10:
Which commit do you use? Can you provide your test code?

FayeY, 2018-05-24:
@berkutta Is there any update? Could you provide a test code so we can reproduce the issue and help debug it?

closed 2018-05-28 by FayeY