espressif/ESP8266_RTOS_SDK #175: Warnings on unsupported toolchain

imi415, 2018-05-07:
The 'Unsupported toolchain' warning is for esp32 toolchain, not esp8266's esp-open-sdk, or maybe I just missed new toolchain update.
```
WARNING: Toolchain version is not supported: 1.22.0-60-g37b07f6
Expected to see version: 1.22.0-80-g6c4433a
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
WARNING: Compiler version is not supported: 
Expected to see version(s): 5.2.0
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
```

27-301, 2018-05-10:
hi, you can just ignore this warning firstly. Since the `make` is just copied from `esp-idf/make`, there should still have some modifications.

FayeY, 2018-05-24:
Closing this, please feel free to reopen if this is still an issue.

closed 2018-05-24 by FayeY