espressif/ESP8266_RTOS_SDK #180: sscanf crash if char* malloc pointer come from second pool

Ymhr6, 2018-05-14:
if my malloc pointer is 0x40108808, it come from second pool (from _lit4_end to 0x4010C000). 
The function sscanf that read on it, will crash with ShowCritical: 0

`C
char* lastSavedTime_ = "2018";
char* lastSavedTime1 = malloc(100);
int p = (int)lastSavedTime1; // p is 0x40108808
strcpy(lastSavedTime1, lastSavedTime_);
int aYear;
sscanf(lastSavedTime1, "%04d", &aYear);
`

Ymhr6, 2018-05-14:
if my malloc pointer is 0x3FFF0350, it come from first pool (from _heap_start to 0x40000000).
The function sscanf will success without problem.

char* lastSavedTime_ = "2018";
	char* lastSavedTime1 = NULL;
	int p = 0;
	while(1)
	{
		lastSavedTime1 = malloc(100);
		p = (int)lastSavedTime1;
		if(p<0x40100000) break; // p is 0x3FFF0350
	}
	strcpy(lastSavedTime1, lastSavedTime_);
	int aYear;
	sscanf(lastSavedTime1, "%04d", &aYear);

FlavioBayer, 2018-05-16:
I was facing an issue while tring to print a number with printf from a created task. It would print the number and (usually) some garbage after (but it never actually crashed).
After reading this issue I did some tests and checked that when the task stack is allocated the garbage is printed only when the task stack is located after 0x4010C000.
I used this code to check what I just said:
`void task_code(void *d){
    printf("-- %x %d\n", (unsigned)&d, (int)d);
    //printf("-- %p %d\n", &d, (int)d); // %p will block and won't print if d>0x4010C000
    vTaskDelay(0xFFFFFFFF);
}
void user_init(void){
    int i;
    char task_name[8];
    for (i=0; i<30; i++){
        sprintf(task_name, "task-%d", i);
        xTaskCreate(task_code, task_name, 512, (void*)i, 1, NULL);
    }
}`
And the output generated:
`-- 3fff3850 0
-- 401081b0॥�� 1॥��
-- 40108a20P���� 2P����
-- 401092a0Х��� 3Х���
-- 40109b10@���� 4@����
-- 4010a390����� 5�����
-- 4010ac000���� 60����
-- 4010b480����� 7�����
-- 4010bcf0 ���� 8 ����
-- 3fff4060 9
-- 3fff4870 10
-- 3fff5080 11
-- 3fff5890 12`

FlavioBayer, 2018-05-16:
As per commit 2e899832829b16fd90a462ed9a2b1bb1070b1ea3 (which introduced iram as the second heap region): "Don't malloc task stack in iram;", which is currently not being enforced (check https://github.com/espressif/ESP8266_RTOS_SDK/blob/master/components/freertos/freertos/tasks.c#L772, it is just a regular malloc that allocate wherever it can). Most likely caused after the recent FreeRTOS update.
This can be fixed by forcing the allocation of task stack to ignore iram, which can be achieve by replacind the line with:
````c
pxStack = ( StackType_t * ) pvPortMalloc( ( ( ( size_t ) usStackDepth ) * sizeof( StackType_t ) ) , __FILE__, __LINE__, false);
```

I'll leave for the developers at Espressif to come up with a final solution, since I'm not an expert when dealing with memory regions of this MCU.
As of now, anyone facing this problem should be able to fix it by doing the change above.

By the way, looks like `MEMLEAK_DEBUG` is defined by default, which add some overhead when using the heap. This may cause issues when `pvPortMalloc` is called with only the allocation size as parameters (which seems to be the case in FreeRTOS, WolfSSL and others, including some precompiled libs in components/esp8266/lib/*.a), since other parameters like `use_iram` would not be defined and may lead to allocation in a random heap region. Shouldn't it be disabled by default, since almost anyone would really use it?

27-301, 2018-05-18:
@Ymhr6 Which commit do you test? I test the commit 99dea43a. It seems OK.

My test code:
```c
void user_init(void)
{
    printf("SDK version:%s\n", system_get_sdk_version());

    char* lastSavedTime_ = "2018";
    char* lastSavedTime1 = malloc(100);
    int p = (int)lastSavedTime1; // p is 0x40108808
    strcpy(lastSavedTime1, lastSavedTime_);
    int aYear;
    sscanf(lastSavedTime1, "%04d", &aYear);

    printf("--->%x %s\n", p, lastSavedTime1);
}
```

My log:
```
rf cal sector: 507
SDK version:2.0.0(06675a3)
--->401079a0 2018
mode : sta(a0:20:a6:04:0d:74)
add if0
scandone
```

27-301, 2018-05-18:
@Ymhr6  Could you please help to do a test.

unzip this file to components/esp8266/source, and have a try. If test OK, we will merge this fix into master.

Thanks

[xtensa_vectors.S.zip](https://github.com/espressif/ESP8266_RTOS_SDK/files/2015958/xtensa_vectors.S.zip)

FlavioBayer, 2018-05-18:
@27-301 I can confirm this bug is affecting the latest commit in release/v2.x.x (581aec06312bf68e2a077e55f887a5eb3221546c).
If I run your test code it works OK, but if I create a task and run it I get "Fatal exception (20)".
Since source code for xtensa_vectors.S was only available to the public after commit 581aec06312bf68e2a077e55f887a5eb3221546c, which is not available for release/v2.x.x, we would need a different approach to test the fix.

27-301, 2018-05-18:
@Ymhr6 We will fix it in branch release/v2.x.x, after that, i will let you know.

27-301, 2018-05-18:
@Ymhr6 Master branch has fixed this issue. Advise you to use the master branch. For old style SDK, it will only be maintained limited, and no new features will be added.

Ymhr6, 2018-05-18:
bo, I don't understand, zip folder structure from master branch and from release 2.0.0 are completely different. I use release zip because this page https://www.espressif.com/en/support/download/sdks-demos point me to release zip.

27-301, 2018-05-19:
@Ymhr6 Just advise you to use master branch, since there are a lot of new features, and bug will be fixed more in time. If you don't work for production.

closed 2018-09-18 by FayeY