espressif/ESP8266_RTOS_SDK #184: STA opmode not work well

c-hamster, 2018-05-17:
in user_init function I do the below code 

- void ICACHE_FLASH_ATTR user_init(void)
- {  UartRxQueue=xQueueCreate(128,sizeof(uint8));//接收队列
-    struct station_config config;
-     wifi_set_opmode(STATION_MODE);
-     printf("nEsp8266 Os is Runing... @%s\r\n",system_update_cpu_freq(160)?"160Mhz":"80Mhz");
-    printf("Ver:%s [%s  %s] UseablRam=%d\n",APP_VER,__DATE__,__TIME__,system_get_free_heap_size());

_- 	memset(&config, 0, sizeof(config));  
- 	sprintf(config.ssid, DEMO_AP_SSID);
- 	sprintf(config.password, DEMO_AP_PASSWORD);
- 	wifi_station_set_config(&config);
- 	wifi_set_event_handler_cb(wifi_handle_event_cb);
- 	wifi_station_connect();_
- 
- //与打印机通信的任务
-    xTaskCreate(UartTask,		"MCU_UART",	255, NULL, 1, NULL);//MCU通信任务	
- //与服务器的通信任务
-   xTaskCreate(ClientTaks,		"AP_con", 512, NULL, 1, NULL);    //连接任务
}

if the ssid and password are right the program will run without any error. as we know 
	wifi_station_set_config(&config); will save the parament to flash, and    when the power is on. It will connect to the AP by default. so I annotate this code. when I power on, the program crashed . finally reset by watchdog. the log is below.

Esp8266 Os is Runing... @160Mhz
Ver:1.000 [May 17 2018  21:58:52] Useable Ram=36344
Chip ID=3e1bd1
mode : sta(5c:cf:7f:3e:1b:d1)
add if0
打印机MCU通信任务开始运行...
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 5
pm open phy_2,type:2 0 0
cnt 

connected with JCG-3F271C, channel 6
dhcp client start...HW (this is my added  imformation in tcpip_adapter.c(line 218) )
      PC: 0x4022e54a        PS: 0x00000033        A0: 0x4022e547        A1: 0x3fffb780  
      A2: 0x3fff38a4        A3: 0x00000003        A4: 0x00000000        A5: 0x00000001  
      A6: 0x4023b9c0        A7: 0x00000000        A8: 0x00000003        A9: 0x00000009  
     A10: 0xfffff000       A11: 0x00000002       A12: 0x00000000       A13: 0x00000000  
     A14: 0x00000001       A15: 0x00000000       SAR: 0x00000013  EXCCAUSE: 0x0000001c  

in earlyer  SDK version ,we can find "user1.1024.new.2.S"file to see where the program crashed.
but I can not find where the file is in this version. 

please help me how to solve this problem! tks a lot!

27-301, 2018-05-18:
@c-hamster Thanks for your report.

We will reproduce this issue.

If you want to know the crash address as old SDK, there are 2 methods can be applied:
1. You can refer to https://esp-idf.readthedocs.io/en/latest/get-started/index.html#monitor , by using `make monitor`, you can get the crash info directly.
2. in your application folder, `cd build`, then `xtensa-lx106-elf-objdump -S xxxx.elf > xxxx.S`, notice: replace xxxx with your application name.

27-301, 2018-05-18:
From your previous post, the crash address is `PC: 0x4022e54a`, now is `4022e58a`? Can you share your *.S here?

c-hamster, 2018-05-18:
because I have rebuild the project.
*.S is not in this compute ,I will send you next monday .

closed 2018-09-18 by FayeY