espressif/ESP8266_RTOS_SDK #187: can i setup a udp server can be accept by a public ip？

fafactx, 2018-05-20:
我想通过esp8266建立一个udp服务器。esp8266 智能配网完毕。sta模式它怎么配置？才能使得我外网服务器可以访问到esp8266？我不想路由器做端口映射。有没有别的办法？

27-301, 2018-05-21:
@fafactx There are no methods in esp8266 side now.

fafactx, 2018-05-28:
已经解决了。外网（公网）IP是无法主动获得局域网内esp8266 IP的，但是esp8266可以通过心跳先发消息给外网IP，这样路由器会有NAT的老化时间（UDP是2min好像）。
6.3.7 配置NAT 地址映射表项老化时间
《华为路由器学习指南》第6章NAT配置与管理，本章主要介绍私网与Internet公网之间的IP地址映射。本节为大家介绍配置NAT 地址映射表项老化时间。
作者：王达来源：人民邮电出版社|2014-08-21 15:07 收藏  分享
人工智能+区块链的发展趋势及应用调研报告

6.3.7 配置NAT 地址映射表项老化时间
这也是一项可选配置任务，可根据需要为NAT 地址映射表配置老化时间，以控制用户对NAT 配置的使用，确保内、外网的通信安全。
配置NAT 地址映射表项老化时间的方法也很简单，只须在系统视图下使用firewall-nat session { dns | ftp | ftp-data | http | icmp | tcp | tcp-proxy | udp | sip | sip-media | rtsp |rtsp-media }aging-time time-value 命令配置即可。参数 time-value的取值范围为1～65 535的整数秒。如果要配置多个会话表项的超时时间需要分别用本命令配置。
缺省情况下，各协议的老化时间为：DNS（120 s）、ftp（120 s）、ftp-data（120 s）、HTTP（120 s）、icmp（20 s）、tcp（600 s）、tcp-proxy（10 s）、udp（120 s）、sip（1 800 s）、sip-media （ 120 s ）、rtsp （ 60 s ）、rtsp-media （ 120 s ）， 可用undo firewall-natsession { all | dns | ftp | ftp-data | http | icmp | tcp | tcp-proxy | udp | sip | sip-media | rtsp |rtsp-media } aging-time 命令恢复对应会话表项的超时时间为缺省值。

closed 2018-05-28 by fafactx