espressif/ESP8266_RTOS_SDK #189: Improve pvPortMalloc and familiy

trygvis, 2018-05-20:
Hi

Right now pvPortMalloc and friends (pvPortCalloc(), pvPortZalloc() and pPortFree()) are not declared in `portable.h` which most of the code expect, but instead they are declared in `lwipopts.h`. This works for a bunch of code because it is included by some esp code (`lwipopts.h` <- `opt.h` <- `ip_addr.h` <- `esp_misc.h`) but fails for a lot of code too.

`MEMLEAK_DEBUG` is used to create some overloaded versions of the functions, but they re-use the same symbol name (to remain compatible with the binary blobs I assume). This is very bad as the extra arguments will contain random garbage from the stack.

27-301, 2018-05-22:
@trygvis Thanks, it really needs some clean up.

trygvis, 2018-05-22:
@27-301 Thanks. I've made a pull request: #192, would appreciate some comments there.

wujiangang, 2018-05-24:
Merged in 72874eb.

closed 2018-05-24 by wujiangang