espressif/ESP8266_RTOS_SDK #199: Flashing firmware using esp-flash-tools_3.6.4

KrrishnaShah, 2018-05-30:
Hello every one,

I'm using virtual_box - eclipse ide - ESP8266_RTOS_SDK.
I've recently updated my sdk with v2.0.0. I've esp8266MOD with 4MB flash.

how can i flash my device using efp-flash-tools v_3.6.4? 

I used to flash esp8266 with previous version like this:
[image]

I tried the same but the program fail to run.
Can any body help me with this ....

27-301, 2018-05-30:
Can you describe the failure more detailed when you use esp-flash-tools v_3.6.4? Can you use old tool to flash your new firmware based on SDK 2.0.0? Can you use the new tool to flash your old firmware?

KrrishnaShah, 2018-05-30:
I'm using this to flash my esp8266

[image]

output:
[image]

27-301, 2018-05-30:
Maybe it's not related to the download tool, from your log, it's already boot up.

Seems that you don't use the original SDK, since you use rboot as bootloader, can you use examples in SDK and espressif boot to do the test?

Since your blinky.bin starts from 0x2000, do you modify the ld file? From you crash info, your PC is 0x40201d74, which means the address in flash is 0x1d74, but 0x1000~0x2000 in your system is blank_config.bin, not your firmware.

Do you use master branch? or just release version?

KrrishnaShah, 2018-05-30:
sry for previous 

now i'm using following boot-loader 
but still fails

[image]

[image]

KrrishnaShah, 2018-05-30:
Thanks @27-301 for guiding me through this.

The problem is solved when I untick the DoNotChgBin.

closed 2018-05-30 by KrrishnaShah