espressif/ESP8266_RTOS_SDK #201: About mqtt MQTTDisconnect

Shaco-Ma, 2018-06-01:
1,In mqtt client, I move ESP8266's mqtt of third_party ,but maybe the example of MQTTEcho.c 's mqtt_client_thread, if first MQTTStartTask ,maybe get the MQTTConnect's ack, and then, server will rst the connection, so MQTTConnect will be fail. I guess MQTTConnect should first init before MQTTStartTask init.
2.The lib of mqtt, this funtion of MQTTDisconnect, if I disconnect the connection Why just MutexUnlock ,the task not do exits?
`int MQTTDisconnect(MQTTClient* c)   
 {  
     int rc = FAILURE;               
     Timer timer;     // we might wait for incomplete incoming publishes to complete
     int len = 0;
    
 #if defined(MQTT_TASK)
     MutexLock(&c->mutex);           
 #endif
     TimerInit(&timer);
     TimerCountdownMS(&timer, c->command_timeout_ms);
        
       len = MQTTSerialize_disconnect(c->buf, c->buf_size);
     if (len > 0)                    
         rc = sendPacket(c, len, &timer);            // send the disconnect packet
     MQTTCloseSession(c);     
 
 #if defined(MQTT_TASK)
       MutexUnlock(&c->mutex);
       //should exit the task? and free all resources?
 #endif
     return rc;
 }
`

27-301, 2018-06-02:
hi @Shaco-Ma can you try master branch? @trygvis just update paho mqtt to newest version and also has some fixes.

trygvis, 2018-06-03:
The MQTT client is not the best designed. For example, as @Shaco-Ma said, the thread should stop when the connection is closed. The thread could also start automatically after a successful connect.

The main problem with these changes is that the code would be harder to maintain in the future if one want to synchronize with upstream's version.

Shaco-Ma, 2018-06-04:
@27-301 I caught the newest code in third_party in release in 2.x.x, but it is update in two mouth ago, however , I repair the code about mqtt in my local,  what version did U say?
Maybe move it to ESP32 is wrong .
Best regareds.

closed 2018-06-04 by Shaco-Ma