espressif/ESP8266_RTOS_SDK #202: cannot assign Static IP

KrrishnaShah, 2018-06-01:
I've upgraded my sdk to 2.0.0 and I'm unable to assign the static IP to my device:

-------------------------------------------------------------------------------------------------
main.c
-------------------------------------------------------------------------------------------------

	#include "FreeRTOS.h"
	#include "task.h"
	#include "esp_common.h"

	#include "driver/gpio.h"
	#include <c_types.h>

	extern void wifi_station_init(void);
	void wifi_station_wait_if_connecting(void);

	void gpio16_output_toggle(void);
	static void __blinky(void * pv);

	void user_init(void)
	{
		printf("\r\n\r\nSDK Version:%s", system_get_sdk_version());
		printf("\r\nESP2866 chip id:0x%x", system_get_chip_id());

		xTaskCreate(__blinky, "__blinky", 200, NULL, 1, NULL);
	}

	static void __blinky(void * pv)
	{
		gpio16_output_conf();

		wifi_station_init();
		wifi_station_wait_if_connecting();

		do
		{
			vTaskDelay(50);
			gpio16_output_toggle();
			printf("\r\nfree heap: %d\r\n RSSI: %d", xPortGetFreeHeapSize(), wifi_station_get_rssi());
		} while(1);
	}

	void gpio16_output_toggle(void)
	{
		static uint32_t state_16 = 0;
		state_16 ^= 1;
		gpio16_output_set(state_16);
	}

	uint32 user_rf_cal_sector_set(void)
	{
		flash_size_map size_map = system_get_flash_size_map();
		uint32 rf_cal_sec = 0;

		switch (size_map) {
			case FLASH_SIZE_4M_MAP_256_256:
				rf_cal_sec = 128 - 5;
				break;

			case FLASH_SIZE_8M_MAP_512_512:
				rf_cal_sec = 256 - 5;
				break;

			case FLASH_SIZE_16M_MAP_512_512:
			case FLASH_SIZE_16M_MAP_1024_1024:
				rf_cal_sec = 512 - 5;
				break;

			case FLASH_SIZE_32M_MAP_512_512:
			case FLASH_SIZE_32M_MAP_1024_1024:
				rf_cal_sec = 1024 - 5;
				break;
			case FLASH_SIZE_64M_MAP_1024_1024:
				rf_cal_sec = 2048 - 5;
				break;
			case FLASH_SIZE_128M_MAP_1024_1024:
				rf_cal_sec = 4096 - 5;
				break;
			default:
				rf_cal_sec = 0;
				break;
		}
		return rf_cal_sec;
	}

-------------------------------------------------------------------------------------------------
wifi_station.c
-------------------------------------------------------------------------------------------------
	#include <stdint.h>
	#include "esp_common.h"
	#include "lwip/ip_addr.h"
	#include "lwip/dns.h"
	#include "FreeRTOS.h"
	#include "task.h"

	void wifi_station_wait_if_connecting(void);

	const static char * ssid = "WCS";
	const static char * pswd = "RTSR&DWCS";
	
	enum {
		I_AM_STATION = 0,
		I_AM_ACCESSPOINT,
	};

	enum {
		STATIC_IP = 0,
		DYNAMIC_IP
	};

	typedef struct {
		struct station_config wifi;
		struct ip_info  _ip;
		uint32_t 		ip_mode;
	} _S_WIFI_STATION_INFO __attribute__((aligned(4)));

	uint32_t wifi_station_init(void)
	{
		_S_WIFI_STATION_INFO _sta;
		
		bzero(&_sta, sizeof(_S_WIFI_STATION_INFO));

		ipaddr_aton("192.168.5.149", &_sta._ip.ip);
		ipaddr_aton("192.168.4.1", &_sta._ip.gw);
		ipaddr_aton("255.255.254.0", &_sta._ip.netmask);

		snprintf((char *)_sta.wifi.ssid, 32, "%s", ssid);
		snprintf((char *)_sta.wifi.password, 32, "%s", pswd);

		printf("\r\n\r\nssid: %s", _sta.wifi.ssid);
		printf("\r\npswd: %s", _sta.wifi.password);

		_sta.wifi.bssid_set = 0;

		_sta.ip_mode = STATIC_IP;
	//	_sta.ip_mode = DYNAMIC_IP;

		printf("\r\nIp mode: %s", _sta.ip_mode == DYNAMIC_IP ? "Dynamic IP" : "Static IP");

		if (wifi_get_opmode() == STATION_MODE)
		{
			if (!wifi_set_opmode(STATION_MODE))
			{
				printf("\r\nFialed to set wifi mode!");
				return 0;
			}
		}

		if (_sta.ip_mode == STATIC_IP)
		{
			printf("\r\nStatic Ip");
			printf("\r\nIP: %s", ipaddr_ntoa(&_sta._ip.ip));
			printf("\r\ngw: %s", ipaddr_ntoa(&_sta._ip.gw));
			printf("\r\nmask: %s", ipaddr_ntoa(&_sta._ip.netmask));
			
			if (wifi_station_dhcpc_status())
			{
				if (!wifi_station_dhcpc_stop())
				{
					printf("\r\nFailed to stop DHCP!");
				}
			}
			wifi_set_ip_info(STATION_IF, &_sta._ip);
			printf("\r\nstation parameter set");
		}
		else
		{
			if(!wifi_station_dhcpc_status())
			{
				printf("\r\n\r\nDHCP is not started. Starting it...\r\n");
				if (!wifi_station_dhcpc_start())
				{
					printf("\r\n\r\nDHCP start failed!\r\n");
					return 0;
				}
			}
		}

		if(!wifi_station_set_config(&_sta.wifi))
		{
			printf("\r\n\r\nFailed to set Station config!\r\n");
			return 0;
		}

		wifi_station_connect();
		return 1;

	}

	extern void gpio16_output_toggle(void);

	void wifi_station_wait_if_connecting(void)
	{
		while(wifi_station_get_connect_status() != STATION_GOT_IP)
		{
			vTaskDelay(10);
			gpio16_output_toggle();
			printf("\r\n.");
		}

		struct ip_info __ip = {{0}, {0}, {0}};

		wifi_get_ip_info(STATION_IF, &__ip);

		printf("\r\n__IP: %s", ipaddr_ntoa(&__ip.ip));
		printf("\r\n__gw: %s", ipaddr_ntoa(&__ip.gw));
		printf("\r\n__mask: %s", ipaddr_ntoa(&__ip.netmask));
	}

-------------------------------------------------------------------------
output
-------------------------------------------------------------------------	
	
	17:39:26.482> 
	17:39:26.546>  ets Jan  8 2013,rst cause:2, boot mode:(3,6)
	17:39:26.546> 
	17:39:26.546> load 0x40100000, len 2592, room 16 
	17:39:26.546> tail 0
	17:39:26.546> chksum 0xf3
	17:39:26.546> load 0x3ffe8000, len 764, room 8 
	17:39:26.546> tail 4
	17:39:26.546> chksum 0x92
	17:39:26.546> load 0x3ffe82fc, len 676, room 4 
	17:39:26.546> tail 0
	17:39:26.546> chksum 0x22
	17:39:26.546> csum 0x22
	17:39:26.546> 
	17:39:26.546> 2nd boot version : 1.7(5d6f877)
	17:39:26.546> SPI Speed : 40MHz
	17:39:26.546> SPI Mode : QIO
	17:39:26.546> SPI Flash Size & Map: 32Mbit(512KB+512KB)
	17:39:26.546> jump to run user1 @ 1000
	17:39:26.546> 
	17:39:26.767> OS SDK ver: 2.0.0(0b78c07) compiled @ May 14 2018 14:11:04phy ver: 1055_1, pp ver: 10.7rf cal sector: 1019
	17:39:26.767> 
	17:39:26.767> SDK Version:2.0.0(0b78c07)
	17:39:26.767> ESP2866 chip id:0x12dda3mode : softAP(ee:fa:bc:12:dd:a3)dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)add if1bcn 100
	17:39:26.767> 
	17:39:26.767> ssid: WCS
	17:39:26.767> pswd: RTSR&DWCS
	17:39:26.767> Static Ip
	17:39:26.767> IP: 192.168.5.149
	17:39:26.767> gw: 192.168.4.1
	17:39:26.767> mask: 255.255.254.0
		  PC: 0x4023965a        PS: 0x00000030        A0: 0x4022e24c        A1: 0x3fff2ac0        A2: 0x00000000        A3: 0x00feffff        A4: 0x3fff2b5c        A5: 0x3fff2b60        A6: 0x3ffe8b82        A7: 0x00000016        A8: 0x6000001c        A9: 0x000000a0       A10: 0x0000000a       A11: 0x00000041       A12: 0x00000000       A13: 0x3fff2b58       A14: 0x3fff2b60       A15: 0x3fff2b5c       SAR: 0x0000001f  EXCCAUSE: 0x0000001d

KrrishnaShah, 2018-06-04:
Can anybody help me with this here ?
I'm desperate ....

FayeY, 2018-06-05:
How about move wifi_station_init(); into user_init
```

void user_init(void)
{
	printf("\r\n\r\nSDK Version:%s", system_get_sdk_version());
	printf("\r\nESP2866 chip id:0x%x", system_get_chip_id());

	wifi_station_init();
	xTaskCreate(__blinky, "__blinky", 200, NULL, 1, NULL);
}
```

KrrishnaShah, 2018-06-06:
I tried and still failed ....

FayeY, 2018-12-20:
Hi, sorry for the inconvenience. We have refactored the SDK to be IDF style, would you like to have a try with our latest SDK?