espressif/ESP8266_RTOS_SDK #206: Build System Documentation Missing

GaryHendrick, 2018-06-02:
make/project.mk assets 
"# See docs/build-system.rst for more details."

There is no such document.  It is unclear how to set up a project to include "include" directories.  The project structure is under-documented.
I am a new ESP8266 user, but a long time embedded system developer with extensive experience using C.  I find the RTOS SDK project almost unapproachable difficult to get started on beyond the most trivial examples.

27-301, 2018-06-04:
hi, @GaryHendrick the documents are lack in this project, but since the project structure is same as [esp-idf](https://github.com/espressif/esp-idf), for documents, you can refer to esp-idf, and also https://esp-idf.readthedocs.io/en/latest/index.html. Finally, esp8266 support will be in esp-idf.

GaryHendrick, 2018-06-04:
Thanks for the info.  That does help me personally, however perhaps the project.mk file should be updated to provide readers with an accurate reference to the documentation.

27-301, 2018-06-05:
Yes, will do. Thanks.

closed 2018-06-05 by 27-301