espressif/ESP8266_RTOS_SDK #207: Unable to determin crash. No stacktrace or useful logs

NZSmartie, 2018-06-03:
I'm using the latest version of master branch (0d718b2) and i'm getting a crash that I can't figure out.
```

 ets Jan  8 2013,rst cause:1, boot mode:(3,6)

load 0x40100000, len 2592, room 16
tail 0
chksum 0xf3
load 0x3ffe8000, len 764, room 8
tail 4
chksum 0x92
load 0x3ffe82fc, len 676, room 4
tail 0
chksum 0x22
csum 0x22

2nd boot version : 1.7(5d6f877)
SPI Speed : 40MHz
SPI Mode : DIO
SPI Flash Size & Map: 32Mbit(1024KB+1024KB)
jump to run user1 @ 1000

OS SDK ver: 2.0.0(bba8ae6) compiled @ May 21 2018 16:51:04
phy ver: 1055_1, pp ver: 10.7

rf cal sector: 1019
w_flash
SDK version:2.0.0(bba8ae6)
mode : sta(18:fe:34:d6:dc:0f)
TCPIP_ADAPTER line: 199 start netif[0]
TCPIP_ADAPTER line: 208 Malloc netif:0
TCPIP_ADAPTER line: 209 Add netif:0
add if0
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 1
pm open phy_2,type:2 0 0
cnt

connected with [REDACTED], channel 6
TCPIP_ADAPTER line: 199 start netif[0]
dhcp client start...
TCPIP_ADAPTER line: 97 dhcpc start
TCPIP_ADAPTER line: 76 dhcpc time(ms): 0
TCPIP_ADAPTER line: 76 dhcpc time(ms): 200
TCPIP_ADAPTER line: 76 dhcpc time(ms): 400
TCPIP_ADAPTER line: 76 dhcpc time(ms): 600
TCPIP_ADAPTER line: 76 dhcpc time(ms): 800
TCPIP_ADAPTER line: 76 dhcpc time(ms): 1000
TCPIP_ADAPTER line: 76 dhcpc time(ms): 1200
TCPIP_ADAPTER line: 76 dhcpc time(ms): 1400
TCPIP_ADAPTER line: 76 dhcpc time(ms): 1600
TCPIP_ADAPTER line: 76 dhcpc time(ms): 1800
TCPIP_ADAPTER line: 76 dhcpc time(ms): 2000
      PC: 0x402254f2        PS: 0x00000033        A0: 0x402254ef        A1: 0x3fff29d0
      A2: 0x3ffed6d4        A3: 0x00000003        A4: 0x00000000        A5: 0x00000001
      A6: 0x4023a1b8        A7: 0x00000000        A8: 0x00000003        A9: 0x00000009
     A10: 0xfffff000       A11: 0x22000000       A12: 0x00000000       A13: 0x00000000
     A14: 0x00000001       A15: 0x00000000       SAR: 0x00000013  EXCCAUSE: 0x0000001c
```

performing a Addr2Line over the above addresses gives: (with invalid addresses omitted)
```
/mnt/c/Users/nzsma/Documents/esp/ESP8266_RTOS_SDK/components/freertos/freertos/queue.c:2332
/mnt/c/Users/nzsma/Documents/esp/ESP8266_RTOS_SDK/components/freertos/freertos/queue.c:2332
/mnt/c/Users/nzsma/Documents/esp/ESP8266_RTOS_SDK/components/esp8266/source/esp_wifi_os_adapter.c:197
```

My project is a copy of https://github.com/espressif/ESP8266_RTOS_SDK/tree/0d718b24b35481eb5614a1bc9481bbb81273b6b0/examples/get-started/project_template

and my `user_main.c` looks like:

```c
void user_init(void)
{
    printf("SDK version:%s\n", system_get_sdk_version());

    wifi_set_opmode(STATION_MODE);

    struct station_config wifi_config = {
        .ssid = "[REDACTED]",
        .password = "[REDACTED]",
    };

    wifi_station_set_config(&wifi_config);
}
```

NZSmartie, 2018-06-04:
Because I don't have enough information it's extremely difficult to locate the cause of this issue. Can the maintainers of this project look at providing a stack trace (like ESP-IDF does on the ESP32)?

FayeY, 2018-10-22:
Would you like to have a try with the laster SDK? It is much stable than before. 
Sorry that we do not have stack trace function, it is due to hardware limitation.
Thanks.

closed 2018-12-20 by FayeY