espressif/ESP8266_RTOS_SDK #211: make: *** No rule to make target `bootloader-clean', needed by `clean'.  Stop.

CaineLew, 2018-06-05:
执行make clean的时候提示make: *** No rule to make target `bootloader-clean', needed by `clean'.  Stop.
可以make，但flash后无法运行
load 0x40100000, len 2592, room 16
0x40100000: vector_entr at xtensa_vectors.o:?

tail 0
chksum 0xf3
load 0x3ffe8000, len 764, room 8
tail 4
chksum 0x92
load 0x3ffe82fc, len 676, room 4
tail 0
chksum 0x22
csum 0x22

2nd boot version : 1.7(5d6f877)
SPI Speed : 40MHz
SPI Mode : QIO
SPI Flash Size & Map: 16Mbit(1024KB+1024KB)
jump to run user1 @ 1000

OS SDK ver: 2.0.0(bba8ae6) compiled @ May 21 2018 16:51:04
phy ver: 1055_1, pp ver: 10.7

rf cal sector: 507
w_flash
@�Z-nP�M
        'γ�     %Y)Q
        �O�
�
 5��)�C
\1      )��     5�NPA�  I�^�q�N9��      )d�     5�P
LZP-CJ�
        ��TZ�
�)��%S�)L�C�    I�TXsAҭ9�))%[
                             5��Y
7)i(LA�Q                         5x�!��KC�
3mN�C��
5)�Z    �\h
          ��
            KN
              1C�V9��!�

OS SDK ver: 2.0.0(bba8ae6) compiled @ May 21 2018 16:51:04
phy ver: 1055_1, pp ver: 10.7

rf cal sector: 507

CaineLew, 2018-06-05:
Sorry,it's my fault.

27-301, 2018-06-05:
hi, @CaineLew We will add bootloader component soon.

closed 2018-06-06 by FayeY