espressif/ESP8266_RTOS_SDK #214: How to enable MBEDTLS_SSL_COOKIE_C

malachib, 2018-06-06:
I notice in `mbedtls/esp_config.h`, `MBEDTLS_SSL_COOKIE_C` is not enable-able by the Kconfig.  Is there a way for me to 'manually' enable it?  I've got my own `mbedtls/esp_config.h` which adds it and does an include_next, but it doesn't seem to be picked up by the mbedtls component itself.  Can anyone offer some guidance?

27-301, 2018-06-07:
@malachib Can you modify the default esp_config.h and Kconfig to let it as a configurable option? then raise a PR? Thanks

malachib, 2018-06-07:
Going smoothly.  Right now in my fork, I have SSL cookie as a discrete option to be enabled or disabled.  I cannot determine 100% if SSL cookie is *only* for DTLS or not.  I think so.  If so, I will lump it in with the already-existing DTLS enable flag.  What do you think?

malachib, 2018-06-08:
So, I could use some guidance.  I'm attempting to bring online the dtls_server demo app (found [here](https://github.com/ARMmbed/mbedtls/blob/development/programs/ssl/dtls_server.c)).  I've done this before in other environments with varying success.  

Before I go into detail, should I ask about it in a new issue here, or would it be better to post on bbs.espressif.com?

27-301, 2018-06-08:
You can ask here. Thanks.

malachib, 2018-06-08:
@27-301 excellent, thank you

closed 2018-06-11 by malachib

malachib, 2018-07-03:
FYI, this feature (in mbedtls) appears broken, as per https://github.com/ARMmbed/mbedtls/issues/1124