espressif/ESP8266_RTOS_SDK #215: ESP8266_RTOS_SDK/components/newlib/newlib/port/syscall.c:48:10: error: conflicting types for '_write_r'

kangks, 2018-06-06:
Using `osx-xtensa-lx106-elf-gb404fb9-2.tar.gz` and `ESP8266_RTOS_SDK` on MacOS (El Capitan), I encountered the following when doing `make all`, did I miss anything in my config?

```
CC build/newlib/newlib/port/syscall.o
~/ESP8266_RTOS_SDK/components/newlib/newlib/port/syscall.c:48:10: error: conflicting types for '_write_r'
 _ssize_t _write_r(struct _reent *r, int fd, void *buf, size_t len)
          ^
In file included from ~/ESP8266_RTOS_SDK/components/freertos/include/freertos/FreeRTOS.h:71:0,
                 from ~/ESP8266_RTOS_SDK/components/newlib/newlib/port/syscall.c:22:
~/ESP8266_RTOS_SDK/components/newlib/newlib/include/reent.h:159:17: note: previous declaration of '_write_r' was here
 extern _ssize_t _write_r _PARAMS ((struct _reent *, int, const void *, size_t));
                 ^
make[1]: *** [newlib/port/syscall.o] Error 1
make: *** [component-newlib-build] Error 2
```

kangks, 2018-06-06:
My bad - I enabled reent from FreeRTOS. Resolving this.

closed 2018-06-06 by kangks

27-301, 2018-06-07:
@kangks If you enable reent, and have this error, it should be an issue.

THKDev, 2018-06-23:
The problem is the missing "const" in the function definition in syscall.c

_ssize_t _write_r(struct _reent *r, int fd, **const** void *buf, size_t len)