espressif/ESP8266_RTOS_SDK #222: DTLS server won't run

malachib, 2018-06-11:
While trying to utilize [this example](https://github.com/ARMmbed/mbedtls/blob/development/programs/ssl/dtls_server.c) I'm encountering a WDT reset during the handshake phase.  With debug logs cranked *way up* we (sometimes) instead get a `mbedtls_ssl_handshake returned -0x4e`

I've just about debug logged all I can think of.  The test project is a marriage of an old ESP8266 multicast 'ping' [example](https://github.com/espressif/esp8266-rtos-sample-code/blob/master/04Protocal/Multicast_Demo/app/user/user_main.c) and the aforementioned dtls_server.  So 95% standard example code.  Project can be found here:

https://bitbucket.org/malachib/playground.esp/src/master/src/PGESP-4/poc1a/

Requirements/Recommendations:

* clone it onto your drive to view it, since I use symlinking for some files.  
* to run it:
  * utilize https://github.com/malachi-iot/ESP8266_RTOS_SDK fork, branch add-mbedtls-feature
  * be sure to copy and configure multicast_config, example can be found in PGESP-6 folder.

Since the project is devoted just to the DTLS task, I've allocated large buffers all the way around.

I have *a lot* of log captures, data, and observations so rather than fill out a 10 page initial post I presume a Q&A format would work better for this.  My theory is this relates to DTLS (not LWIP) fragmentation breaking down somehow.

27-301, 2018-06-11:
hi, @malachib Thanks to report this. Since now, the guys are busy refactoring the SDK, this issue will not be investigated immediately, but we've marked it. If you have any update, please also post here. Thanks.

malachib, 2018-07-03:
FYI these issues suggest that mbedtls is not fully functional in this department:

https://github.com/ARMmbed/mbedtls/issues/1124
https://github.com/ARMmbed/mbedtls/issues/387 ( this issue 387 appears to conflate a 'max fragment' option with *any* fragmenting, so it's kind of confusing)

malachib, 2018-07-04:
Further findings.  After seriously intense trial and error, the call to ‘esp_aio_sendto’ *seems* to be what's crashing

malachib, 2018-07-11:
ESP32 environment with identical code (as much as possible) does run as expected.  Am able to connect and transfer data with `OpenSSL s_client`

This includes the expected lack of DTLS Handshake Fragmentation, thus necessitating LwIP IP Fragmentation.  Without this, ESP32 behaves similarly to ESP8266 and doesn't sent the > 1500byte handshake packets.  Not ideal, but is a useful data point.