espressif/ESP8266_RTOS_SDK #224: lwip raw UDP sendto not working as expected

malachib, 2018-06-12:
Using lwip raw udp API, notice that only one UDP echo works then the device  echoes back improperly after that (UDP packet seems to be malformed, missing payload)

``` c++
#include "lwip/api.h"
#include "lwip/udp.h"

void udp_echo_recv(void *arg, 
    struct udp_pcb *pcb, struct pbuf *p,  
    const ip_addr_t *addr, u16_t port)
{
    if (p != NULL) {
        /* send received packet back to sender */
        udp_sendto(pcb, p, addr, port);
        /* free the pbuf */
        pbuf_free(p);
    }
}

void udp_echo_init(void)
{
    struct udp_pcb * pcb;

    /* get new pcb */
    pcb = udp_new();
    if (pcb == NULL) {
        LWIP_DEBUGF(UDP_DEBUG, ("udp_new failed!\n"));
        return;
    }

    /* bind to any IP address on port 7 */
    if (udp_bind(pcb, IP_ADDR_ANY, 7) != ERR_OK) {
        LWIP_DEBUGF(UDP_DEBUG, ("udp_bind failed!\n"));
        return;
    }

    /* set udp_echo_recv() as callback function
       for received packets */
    udp_recv(pcb, udp_echo_recv, NULL);
}
```

tcpdump after asking to echo '123':

```
00:09:21.434673 IP (tos 0x0, ttl 64, id 11414, offset 0, flags [none], proto UDP (17), length 32)
    192.168.2.2.49738 > 192.168.2.179.echo: [udp sum ok] UDP, length 4
	0x0000:  5ccf 7f3a 08ba 58b0 3561 d381 0800 4500  \..:..X.5a....E.
	0x0010:  0020 2c96 0000 4011 c831 c0a8 0202 c0a8  ..,...@..1......
	0x0020:  02b3 c24a 0007 000c 5342 3132 330a       ...J....SB123.
00:09:21.636659 IP truncated-ip - 4 bytes missing! (tos 0x0, ttl 255, id 5, offset 0, flags [none], proto UDP (17), length 32)
    192.168.2.179.echo > 192.168.2.2.49738: UDP, length 4
	0x0000:  58b0 3561 d381 5ccf 7f3a 08ba 0800 4500  X.5a..\..:....E.
	0x0010:  0020 0005 0000 ff11 35c2 c0a8 02b3 c0a8  ........5.......
	0x0020:  0202 0007 c24a 000c 5342  
```

Full code [here](https://bitbucket.org/malachib/playground.esp/src/master/src/PGESP-12/poc1/) 

Any thoughts?

malachib, 2018-06-20:
If I pbuf_alloc/pbuf_copy to a new pbuf and send that, things do work.  Perhaps the incoming pbuf has the wrong kind of header allocated for the udp_sendto?

Junhao-Espressif, 2018-08-17:
Hi @malachib 
Can you try this with release/v3.0 ? Thanks!

malachib, 2018-08-21:
Willdo!

malachib, 2018-09-05:
I tried release/v3.0 SHA b461d81844471d6e40f9e1dbc4df7dc11f7276c7
Unfortunately exact same results
Updated code (to v3.0 APIs) is [here](https://bitbucket.org/malachib/playground.esp/src/master/src/PGESP-12/poc1-3.0/)

malachib, 2018-09-11:
Ah you'll need to do `git init submodule` and `git update submodule` - If that doesn't help I think I can #ifdef the estd stuff out

Junhao-Espressif, 2018-09-12:
Hi @malachib 

I have tested and have not reproduced this problem.

Could you use wireshark or omnipeek to capture packets？

Thank you!

```
11:47:02.769376 IP (tos 0x0, ttl 64, id 24157, offset 0, flags [DF], proto UDP (17), length 34)
    192.168.3.150.39281 > 192.168.3.153.echo: [bad udp cksum 0x889f -> 0x265b!] UDP, length 6
	0x0000:  4500 0022 5e5d 4000 4011 53ee c0a8 0396  E.."^]@.@.S.....
	0x0010:  c0a8 0399 9971 0007 000e 889f 5342 3132  .....q......SB12
	0x0020:  330a                                     3.
11:47:02.772754 IP (tos 0x0, ttl 255, id 3, offset 0, flags [none], proto UDP (17), length 34)
    192.168.3.153.echo > 192.168.3.150.39281: [udp sum ok] UDP, length 6
	0x0000:  4500 0022 0003 0000 ff11 3348 c0a8 0399  E.."......3H....
	0x0010:  c0a8 0396 0007 9971 000e 265b 5342 3132  .......q..&[SB12
	0x0020:  330a 0000 0000 0000 0000 0000 0000       3.............
```

malachib, 2018-09-12:
Just doublechecking, did you try pinging twice?  First one works properly

Junhao-Espressif, 2018-09-12:
Hi @malachib 

```
python udpecho.py -c 192.168.3.153 7
udp echo client ready, reading stdin
123456
client received '123456\n' from ('192.168.3.153', 7)
123456
client received '123456\n' from ('192.168.3.153', 7)
2222223
client received '2222223\n' from ('192.168.3.153', 7)
444444
client received '444444\n' from ('192.168.3.153', 7)
22222
client received '22222\n' from ('192.168.3.153', 7)
123456
client received '123456\n' from ('192.168.3.153', 7)
654321
client received '654321\n' from ('192.168.3.153', 7)
```

```
 ping 192.168.3.153
PING 192.168.3.153 (192.168.3.153) 56(84) bytes of data.
64 bytes from 192.168.3.153: icmp_seq=1 ttl=255 time=3.60 ms
64 bytes from 192.168.3.153: icmp_seq=2 ttl=255 time=6.56 ms
64 bytes from 192.168.3.153: icmp_seq=3 ttl=255 time=5.02 ms
64 bytes from 192.168.3.153: icmp_seq=4 ttl=255 time=3.26 ms
64 bytes from 192.168.3.153: icmp_seq=5 ttl=255 time=3.91 ms
^C
--- 192.168.3.153 ping statistics ---
5 packets transmitted, 5 received, 0% packet loss, time 4005ms
rtt min/avg/max/mdev = 3.262/4.474/6.567/1.205 ms
```

malachib, 2018-09-12:
OK, I will check again.  Maybe something's odd with my sdkconfig

FayeY, 2018-09-18:
Is there any update?

malachib, 2018-09-18:
Not yet.  I do plan to test again shortly.  Been working on an esp32 project.  Perhaps it's a toolchain thing, because I see

```
WARNING: Toolchain version is not supported: 1.22.0-60-g37b07f6
Expected to see version: 1.22.0-88-gde0bdc1
```

Not to hijack my own thread, but I figured using pfalcon's latest `esp-open-sdk` would be right but apparently not

malachib, 2018-09-18:
Attempted again and met with same results.  On the notion of it being a toolchain difference, I tried with the downloaded toolchain (suffixed in -88), but am then met with a different set of errors:

```
make[1]: Entering directory '/home/malachi/Projects/ext/ESP8266_RTOS_SDK_newest/tools/kconfig'
cc -c  -D_GNU_SOURCE -D_DEFAULT_SOURCE -DCURSES_LOC="<ncurses.h>" -DLOCALE -MD  /home/malachi/Projects/ext/ESP8266_RTOS_SDK_newest/tools/kconfig/mconf.c -o mconf.o
as: unrecognized option '--64'
Makefile:171: recipe for target 'mconf.o' failed
make[1]: *** [mconf.o] Error 1
make[1]: Leaving directory '/home/malachi/Projects/ext/ESP8266_RTOS_SDK_newest/tools/kconfig'
make: *** No rule to make target '/home/malachi/Projects/ext/ESP8266_RTOS_SDK_newest/tools/kconfig/conf', needed by '/home/malachi/Projects/playground/playground.esp/src/PGESP-12/poc1-3.0/build/include/config/auto.conf'.  Stop.
```

Being that you are able to compile it, I except I did something dumb, but can't figure out what

FayeY, 2018-10-22:
Could you call "make clean" in the "kconfig" folder, and then compile it again?

malachib, 2018-10-24:
OK I will try that

malachib, 2018-12-24:
Hmm strange.  Very sure I posted an update way back when.  Will have to try it again

FayeY, 2019-01-09:
Closing due to lack of response, please feel free to reopen if this is still an issue or there is any update. Thanks.

closed 2019-01-09 by FayeY