espressif/ESP8266_RTOS_SDK #225: 关于 mqtt demo 重连内存问题

huangliqun, 2018-06-12:
我使用mqtt demo, 写了一个测试程序，连接mqtt服务器后，断开连接，然后再次去连接服务器。
但是发现每次新连接服务器后，可用的内存一直在减少。我关闭连接用的是
**
MQTTDisconnect(&Mqtt_client);
close(network.my_socket);
**

主循环是这样的：
**    while(1) //已经连上路由
    	{
    		NetworkInit(&network);
    		MQTTClientInit(&Mqtt_client, &network, 30000, sendbuf, sizeof(sendbuf), readbuf, sizeof(readbuf));

    		 if ((rc = NetworkConnect(&network, address, MQTT_PORT)) != 0)//连接MQTT TCP
			{
				printf("NetworkConnect err is %d\n", rc);
			}
			else if((rc = MQTTConnect(&Mqtt_client, &connectData)) != 0)//连接MQTT 登录
                      {	
		        printf("MQTTConnect err is %d\n", rc);
		    }
			else if((rc = MQTTSubscribe(&Mqtt_client, "ESP8266/sample/sub", 2, messageArrived)) != 0)                      {
				printf("MQTTSubscribe err is %d\n", rc);
		       }
			else{
				printf("MQTTSubscribe  .ESP8266/sample/sub\n");
			}

		    vTaskDelay(100);
		    os_printf("system heap is: %d ..........\r\n",system_get_free_heap_size());
		    os_printf("mqtt clind to ded: %d ..........\r\n");

		    	MQTTDisconnect(&Mqtt_client);
		    	close(network.my_socket);

    	}**

最后串口打印，可用内存在一直减少：
MQTTSubscribe  .ESP8266/sample/sub
system heap is: 33744 ..........
mqtt clind to ded: 0 ..........
MQTTSubscribe  .ESP8266/sample/sub
system heap is: 33648 ..........
mqtt clind to ded: 0 ..........
MQTTSubscribe  .ESP8266/sample/sub
system heap is: 33552 ..........
mqtt clind to ded: 0 ..........

xcguang, 2018-06-15:
Hi, 请参考如下实现
```c
static void mqtt_client_thread(void* pvParameters)
{
    printf("mqtt client thread starts\n");
    MQTTClient client;
    Network network;
    static unsigned char sendbuf[512], readbuf[512] = {0};
    int rc = 0, count = 0;
    MQTTPacket_connectData connectData = MQTTPacket_connectData_initializer;
    bool first_flag = false;
    
    pvParameters = 0;

    if(DEFAULT_SECURITY == NO_TLS) {
        NetworkInit(&network);
    } else {
        NetworkInitSSL(&network);
    }

    MQTTClientInit(&client, &network, 10000, sendbuf, sizeof(sendbuf), readbuf, sizeof(readbuf));

    for (;;) {
        while ( wifi_station_get_connect_status() != STATION_GOT_IP) {
            vTaskDelay(1000 / portTICK_RATE_MS);
        }
        char* address = MQTT_BROKER;

        printf("begin\n");
        switch(DEFAULT_SECURITY) {
        case NO_TLS:
            if ((rc = NetworkConnect(&network, address, MQTT_PORT)) != 0) {
                printf("Return code from network connect is %d\n", rc);
            }
            break;

        case TLS_WITHOUT_AUTHENTICATION:
            SSL_CA_CERT_KEY_INIT(&ssl_cck, NULL, 0, NULL, 0, NULL, 0);
            if ((rc = NetworkConnectSSL(&network, address, MQTT_PORT, &ssl_cck, TLS_client_method(), SSL_VERIFY_NONE, 8192)) != 1) {
                   printf("Return code from network connect ssl is %d\n", rc);
                }
            break;

        case ONE_WAY_ANTHENTICATION:
            SSL_CA_CERT_KEY_INIT(&ssl_cck, ca_crt, ca_crt_len, NULL, 0, NULL, 0);
            if ((rc = NetworkConnectSSL(&network, address, MQTT_PORT, &ssl_cck, TLS_client_method(), SSL_VERIFY_PEER, 8192)) != 1) {
                   printf("Return code from network connect ssl is %d\n", rc);
                }
            break;

        case TWO_WAY_ANTHENTICATION:
            SSL_CA_CERT_KEY_INIT(&ssl_cck, ca_crt, ca_crt_len, client_crt, client_crt_len, client_key, client_key_len);
            if ((rc = NetworkConnectSSL(&network, address, MQTT_PORT, &ssl_cck, TLS_client_method(), SSL_VERIFY_PEER, 8192)) != 1) {
                   printf("Return code from network connect ssl is %d\n", rc);
                }
            break;
        default:

            break;
        }
        
        connectData.MQTTVersion = 3;
        connectData.clientID.cstring = "ESP8266_sample";
        connectData.username.cstring = MQTT_USER_NAME;
        connectData.password.cstring = MQTT_USER_PASSWORD;
        printf("MQTT Connecting\n");
        if ((rc = MQTTConnect(&client, &connectData)) != 0) {
            printf("Return code from MQTT connect is %d\n", rc);
            network.disconnect(&network);
            vTaskDelay(1000 / portTICK_RATE_MS);
            continue;
        } 
        printf("MQTT Connected\n");

#if defined(MQTT_TASK)
        if (first_flag == false) {
            if ((rc = MQTTStartTask(&client)) != pdPASS) {
                printf("Return code from start tasks is %d\n", rc);
            } else {
                printf("Use MQTTStartTask\n");
                first_flag = true;
            }
        }
#endif
        
        if ((rc = MQTTSubscribe(&client, MQTT_SUB_TOPIC, 2, messageArrived)) != 0) {
            printf("Return code from MQTT subscribe is %d\n", rc);
            network.disconnect(&network);
            vTaskDelay(1000 / portTICK_RATE_MS);
            continue;
        }
        printf("MQTT subscribe to topic %s\n",MQTT_SUB_TOPIC);

        for (;;) {
            MQTTMessage message;
            esp_mqtt_msg_type msg;

            xQueueReceive(esp_mqtt_queue, &msg, portMAX_DELAY);
            
            message.qos = QOS1;
            message.retained = 0;
            message.payload = msg.data;
            message.payloadlen = msg.len;

            if ((rc = MQTTPublish(&client, MQTT_PUB_TOPIC, &message)) != 0) {
                printf("Return code from MQTT publish is %d\n", rc);
            } else {
                printf("MQTT publish topic %s, message: %s\n", MQTT_PUB_TOPIC, message.payload);
            }
            
            free(msg.data);
            
            if (rc != 0) {
                break;
            }
        }
        network.disconnect(&network);
    }
    
    printf("mqtt_client_thread going to be deleted\n");
    vTaskDelete(NULL);
    return;
}
```

27-301, 2018-06-15:
@huangliqun Please don't call MQTTClientInit in the loop, it SHOULD be called only once. It's a limitation in paho mqtt. Thanks

JeremyHe-cn, 2018-07-26:
@27-301 这不是 paho mqtt 的问题好吗。明显是 MQTTClientInit 的问题，麻烦仔细检查对象的申请与释放。要么在 MQTTDisconnect 里释放，要么提供对应 的 deInit 方法

27-301, 2018-07-27:
`MQTTClientInit` & `MQTTDisconnect ` 都是 paho mqtt 库里的，paho mqtt 也没提供 deinit 的方法。

9crk, 2018-08-02:
@27-301 @JeremyHe-cn @xcguang @JeremyHe-cn  你们好！我在使用RTOS SDK里的MQTT例子连接亚马逊aws iot，发现SSL_Connect返回0，无法连接上，但是自己用mosquitto搭建的带SSL的服务器是可以连上的。代码什么都没改，就改了3个认证文件和server地址， 这可能是那个地方的问题？代码流程跟以上代码是一样的。

9crk, 2018-08-02:
使用MQTT.fx ，不管是aws iot还是自己搭建的mosquitto，都可以连上。

27-301, 2018-08-03:
建议你用 master 分支，examples/protocols/aws_iot 下直接有例子。使用前看下 README。

closed 2018-09-18 by FayeY