espressif/ESP8266_RTOS_SDK #227: Fail to auto reconnect to access point ("LWIP_ERROR: netif is not up, old style port?")

DKrepsky, 2018-06-12:
Description:
Esp module fails to auto-reconnect to an AP when auto-reconnect is enabled.

SDK version: 
Current master branch.

Steps to reproduce:
Use any of the station examples to connect to an AP, then reset your router. The esp module should be able to auto-reconnect to the wifi when wifi is ready.

Log message on serial port:
LWIP_ERROR: netif is not up, old style port?

Notes: 
It seems that the DHCP client fails to start and acquire an IP because the netif is down.

27-301, 2018-06-15:
I tried, but not reproduce.

THKDev, 2018-06-18:
I can confirm this issue on branch [master](https://github.com/espressif/ESP8266_RTOS_SDK/tree/1c10af9402489a851938a882e11dee93642064ee). 

```
bcn_timout,ap_probe_send_start
ap_probe_send_over, rest wifi status to disassoc
state: 5 -> 0 (1)
rm 0
pm close 7 0 0/35441728

reconnect
scandone
no NAME found, reconnect after 1s

reconnect
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 3
pm open phy_2,type:2 0 0
cnt

connected with NAME, channel 11
dhcp client start...
netif is not up, old style port?
Assertion: 727 in /home/user/projects/esp/ESP8266_RTOS_SDK/components/lwip/lwip/src/core/ipv4/dhcp.c                                                                  
ShowCritical:0
HdlMacSig:0
SWReq:0
```

DKrepsky, 2018-06-18:
I've tracked down the issue and it happens as follow:

- After the wifi signal is lost, function _tcpip_adapter_sta_leave_ is called. This will shutdown the interface esp_netif[TCPIP_ADAPTER_IF_STA].

- When the wifi connection is re-established, function _tcpip_adapter_start_ is called, which tries to start the DHCP server by passing the previously released interface. This will cause the error, since esp_netif[TCPIP_ADAPTER_IF_STA] is not a valid interface anymore.

I've fixed it by adding a check to verify if the interface is up. If not, it will be started before calling tcpip_adapter_station_dhcp_start().

THKDev, 2018-06-18:
Confirmed. Such changes will solve this issue. A sample patch is attached.
[patch.zip](https://github.com/espressif/ESP8266_RTOS_SDK/files/2112411/patch.zip)

27-301, 2018-06-19:
hi, @DKrepsky @THKDev Thanks! Can you raise a PR?

27-301, 2018-07-10:
@DKrepsky Can you check this by using the latest master?

closed 2018-10-22 by FayeY