espressif/ESP8266_RTOS_SDK #228: Does any mechanism replace ICACHE_RAM_ATTR?

cranphin, 2018-06-14:
In the non-os version of the SDK, you use ICACHE_RAM_ATTR for interrupt handlers and such, which I think makes the code run from/load in ram instead of flash?

The same marker seems to not be in the RTOS SDK, and I don't see any markers on interrupts in the code.

Why is this, and is there still anything I should be aware of? :)

Thank you!

xcguang, 2018-06-15:
Hi,
Please try `IRAM_ATTR`.

27-301, 2018-06-15:
@cranphin You can put some critical code in iRAM by using IRAM_ATTR manually, but now you don't need to be aware of the interrupt code in RTOS SDK. The interrupt code can be in flash also. For `IRAM_ATTR`, you need to include "esp_attr.h" firstly. Thanks.

cranphin, 2018-06-15:
Awesome, thank you :)

closed 2018-06-15 by cranphin