espressif/ESP8266_RTOS_SDK #230: New bootloader stuck in endless loop

THKDev, 2018-06-18:
## Environment

- Development Kit: ???
- IDF version: 1be2401
- Development Env:     Make
- Operating System:     Fedora
- Power Supply:         external 3.3V

## Problem Description
New bootloader runs in a endless loop, because of missing init data.

### Expected Behavior
Should boot up as normal.

### Actual Behavior
endless loop.

### Steps to repropduce

1. make erase_flash  (!! important !!)
2. make flash
3. reset ESP8266

### Code to reproduce this issue
any example

## Debug Logs

```
load 0x40108000, len 704, room 16
tail 0
chksum 0xe0
load 0x3ffe8000, len 764, room 8
tail 4
chksum 0x81
load 0x3ffe82fc, len 380, room 4
tail 8
chksum 0x2d
csum 0x2d

2nd boot version : 2.0
  SPI Speed      : 40MHz
  SPI Mode       : QIO
  SPI Flash Size & Map: 16Mbit(1024KB+1024KB)
jump to run @ 10000

OS SDK ver: 2.0.0(fd9b17b) compiled @ May 31 2018 14:03:04
rf_cal[0] !=0x05,is 0xFF

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)
```

## Other items if possible

- [x] sdkconfig file (attach the sdkconfig file from your project folder)

[sdkconfig.zip](https://github.com/espressif/ESP8266_RTOS_SDK/files/2112091/sdkconfig.zip)

DKrepsky, 2018-06-18:
I believe you need to erase the entire flash and then write the SDK init data at a specified address.

The current documentation is lacking this, but it is well explained in the nodeMCU docs:
[SDK Init Data](http://msr-nodemcu.readthedocs.io/en/dev-rtd-test/en/flash/#sdk-init-data)

closed 2018-06-19 by wujiangang

27-301, 2018-06-19:
@THKDev @DKrepsky flash map still under refactor, and it will be finally similar to idf. We will have a legacy mode which will have init data in (last - 3) sector, but in normal mode, phy init data will be in 0xf000 or just in code.