espressif/ESP8266_RTOS_SDK #231: SDK 2.0 SPI Master issue

berkutta, 2018-06-18:
I'm using the newest 2.0 SDK and facing issues with the SPI Master example from https://github.com/espressif/esp8266-rtos-sample-code/blob/master/02Peripheral/HSPI_Master/user/user_main.c

Cyan/Blue is the CLK Line (GPIO14). The signal is very unstable and noisy. My Slave SPI Device (SSD1331 Display) can't handle a such noisy clock signal. Is there a solution for that?

[image]

xiongyumail, 2018-09-20:
Hi @berkutta 
Maybe you need an external pull-up resistor?

berkutta, 2018-10-11:
I have now tried with a 10k Pullup Resistor on HSPI CLK but it's still nearly unusuable.

berkutta, 2019-01-16:
Looks like it got fixed with the new SPI driver. Thank you Espressif!

[image]

xiongyumail, 2019-01-17:
Thank you!
I think it might be related to this option.
`CLEAR_PERI_REG_MASK(PERIPHS_IO_MUX_CONF_U, SPI1_CLK_EQU_SYS_CLK);`

closed 2019-01-21 by FayeY