espressif/ESP8266_RTOS_SDK #234: Error when build and upload by "make flash"

TridentTD, 2018-06-23:
## Environment

- Development Kit: Wemos D1 Mini
- Development Env:      Arduino IDE
- Operating System:     Windows
- Power Supply:         USB

## Problem Description
After 'make menuconfig' to edit Serial COM port.
The 'project_template' can be built  by calling "make flash" ,
but when uploading step by "make flash"
it's occur the error as the following

[image]

How to fix this problem ?

27-301, 2018-06-24:
hi, can you check that your serial port is COM4? It seems that esptool cannot open COM4.

TridentTD, 2018-06-25:
When I use  Arduino-Platform's esptool.exe,
it can open COM4 and can upload, 
but  esptool.py  by 'make flash' occurs error.

27-301, 2018-06-27:
@projectgus Do you have any idea about this? Thanks

projectgus, 2018-07-02:
Hi @TridentTD,

Is it possible there is some security software or something similar running on your PC which might be blocking access? You can try running the MINGW32 shell with "Run as Administrator" and see if this changes anything.

If this doesn't help, where did you install MINGW32 from? Are you using the ESP-IDF MINGW32 environment, or a different one? Is it possible you have a Cygwin version of Python not mingw32 (you can check by running `python -c "import sys; print(sys.platform)"`, it should say `nt`).

wanghui0574xs, 2018-09-01:
make flash
WARNING: Toolchain version is not supported: 1.22.0-60-g37b07f6
Expected to see version: 1.22.0-88-gde0bdc1
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
Please download and use the toolchain from the URL of README.md
WARNING: Toolchain version is not supported: 1.22.0-60-g37b07f6
Expected to see version: 1.22.0-88-gde0bdc1
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
Please download and use the toolchain from the URL of README.md
Flashing binaries to serial port /dev/tty.SLAB_USBtoUART (app at offset 0x10000)...
esptool.py v2.4.0
Traceback (most recent call last):
  File "/Users/wanghui/Workspace/GitProject/ESP8266/ESP8266_RTOS_SDK/components/esptool_py/esptool/esptool.py", line 3022, in <module>
    _main()
  File "/Users/wanghui/Workspace/GitProject/ESP8266/ESP8266_RTOS_SDK/components/esptool_py/esptool/esptool.py", line 3015, in _main
    main()
  File "/Users/wanghui/Workspace/GitProject/ESP8266/ESP8266_RTOS_SDK/components/esptool_py/esptool/esptool.py", line 2711, in main
    ser_list = sorted(ports.device for ports in list_ports.comports())
  File "/Users/wanghui/Workspace/GitProject/ESP8266/ESP8266_RTOS_SDK/components/esptool_py/esptool/esptool.py", line 2711, in <genexpr>
    ser_list = sorted(ports.device for ports in list_ports.comports())
AttributeError: 'list' object has no attribute 'device'
make: *** [flash] Error 1

27-301, 2018-09-14:
@wanghui0574xs Please check the dev board is connected to you Mac, you can use `ls /dev/tty.*` to check.

closed 2018-09-20 by FayeY

jiekechoo, 2018-10-23:
@wanghui0574xs You could update "pyserial" to latest version. 
`pip uninstall pyserial`
`pip install pyserial`