espressif/ESP8266_RTOS_SDK #235: Memory leak

THKDev, 2018-06-23:
[memleak.zip](https://github.com/espressif/ESP8266_RTOS_SDK/files/2129817/memleak.zip)
## Environment

- Development Kit: ESP12-E
- IDF version c7c96e00660d8ce236ebac606786b9d064a2d03a
- Development Env:      Make
- Operating System:     Fedora (Linux)
- Power Supply:         external 3.3V

## Problem Description
Available heap memory is shrinking more and more as longer the program is running. Maybe a leak inside the FreeRTOS port. 
Another leak can be triggered by switching the wifi router off and switching on again. After the ESP8266 has reconnected the size of available heap has been decreased.

### Expected Behavior
No memory leaks.

### Actual Behavior
Has some memory leaks.

### Steps to repropduce
1. adjust SSID and password inside main/wifi.c
2. compile and flash attached sample 
3. attach serial monitor
4. let it run for 10 min - see first memory leaks
5. switch wifi router off
6. switch wifi router on
7. see some more memory leaks

### Code to reproduce this issue
See attachement.

## Debug Logs
See attachement.

## Other items if possible

- [x] sdkconfig file (attach the sdkconfig file from your project folder)

THKDev, 2018-06-24:
Some more details. After a wifi reconnect 5 buffer are left which where allocated at https://github.com/espressif/ESP8266_RTOS_SDK/blob/305a338a27eb4d0cf0b360d8246bd5c685bfb037/components/lwip/lwip/src/core/ipv4/etharp.c#L1115
It seems that the refcounter is wrong an so the pbuf is NOT released. https://github.com/espressif/ESP8266_RTOS_SDK/blob/305a338a27eb4d0cf0b360d8246bd5c685bfb037/components/lwip/lwip/src/core/ipv4/etharp.c#L1162
Maybe the call to https://github.com/espressif/ESP8266_RTOS_SDK/blob/305a338a27eb4d0cf0b360d8246bd5c685bfb037/components/lwip/port/esp8266/netif/ethernetif.c#L142
and refcounting pbuf at https://github.com/espressif/ESP8266_RTOS_SDK/blob/305a338a27eb4d0cf0b360d8246bd5c685bfb037/components/lwip/port/esp8266/netif/ethernetif.c#L91 is the reason for this leak.

[First connect to wifi heap dump](https://github.com/espressif/ESP8266_RTOS_SDK/files/2130621/heap1.txt)
[Seccond connect wifi heap dump](https://github.com/espressif/ESP8266_RTOS_SDK/files/2130622/heap4.txt)

27-301, 2018-07-10:
@THKDev Can you check with the latest master? Thanks.

THKDev, 2018-07-10:
I would like but there are to many missing functions declarations like `wifi_station_disconnect, wifi_station_disconnect, ...` . For more details see attached log file. Did i miss something ?

[log.zip](https://github.com/espressif/ESP8266_RTOS_SDK/files/2181505/log.zip)

cranphin, 2018-07-10:
It looks like wifi got refactored, look at the protocol examples to see how wifi should be initialized now :)

THKDev, 2018-07-11:
@cranphin Thank you!
@27-301 Example [wifi station machine](https://github.com/espressif/ESP8266_RTOS_SDK/blob/cb0c5ee5fce091b03a2af025f4bc59e8c188d0f0/examples/wifi/wifi_station_machine) has to be fixed to fit the new API.

THKDev, 2018-07-15:
Seems that some leaks sealed. But there still some remaining. Test started with 48688 bytes of free heap. After 18 hour 47552 bytes remaining. [See log with heap dump](https://github.com/espressif/ESP8266_RTOS_SDK/files/2195599/heap.zip) .

Junhao-Espressif, 2018-08-17:
Hi @THKDev 
Can you check with release/v3.0 ? Thanks!

FayeY, 2018-10-22:
Closing due to lack of response, please feel free to reopen if this is still an issue. Thanks.

closed 2018-10-22 by FayeY