espressif/ESP8266_RTOS_SDK #236: Why there are so many syntax error in version 2.0?Is not a stable version?

aeris-ming, 2018-06-23:
/home/aeris/ESP-8266-TIM/ESP8266_RTOS_SDK/driver_lib/include/uart.h:104:5: error: enumerator value for 'UART_Rxd_Inverse' is not an integer constant
     UART_Rxd_Inverse  = (const uint32_t)UART_RXD_INV,
     ^
1.Why not use uint8_t uint32_t ...?!!!
2.
/home/aeris/ESP-8266-TIM/ESP8266_RTOS_SDK/driver_lib/include/uart.h:105:5: error: enumerator value for 'UART_CTS_Inverse' is not an integer constant
     UART_CTS_Inverse  = (const uint32_t)UART_CTS_INV,
     ^
/home/aeris/ESP-8266-TIM/ESP8266_RTOS_SDK/driver_lib/include/uart.h:106:5: error: enumerator value for 'UART_Txd_Inverse' is not an integer constant
     UART_Txd_Inverse  = (const uint32_t)UART_TXD_INV,
     ^
/home/aeris/ESP-8266-TIM/ESP8266_RTOS_SDK/driver_lib/include/uart.h:107:5: error: enumerator value for 'UART_RTS_Inverse' is not an integer constant

27-301, 2018-06-24:
hi, uart driver is under refactoring now, if it's available, i will ping you. Thanks

FayeY, 2018-09-20:
The UART driver has been refactored in V3.0, please have a try. Thanks.

closed 2018-09-20 by FayeY