espressif/ESP8266_RTOS_SDK #237: Instant crash since recent update

cranphin, 2018-06-23:
If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

2. Describe your problem.
3. Include [debug logs on the monitor](http://esp-idf.readthedocs.io/en/latest/get-started/idf-monitor.html#automatically-decoding-addresses).
4. Provide more items under **Other items if possible** can help us better locate your problem.
5. Use markup (buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

----------------------------- Delete above -----------------------------

## Environment

- Development Kit: ESP8266_RTOS_SDK
- IDF version (``git rev-parse --short HEAD`` to get the commit id.): 
    305a338a27eb4d0cf0b360d8246bd5c685bfb037 
- Development Env:      Make
- Operating System:     Windows
- Power Supply:         USB

## Problem Description

I have done a update of the SDK (latest from GIT) which was several weeks old.
Now the chip instantly crashes when powered on.
Even with a very basic program. I've added the serial output.

### Expected Behavior

No crashing, like before the update of the SDK

### Actual Behavior

See serial output log

### Steps to repropduce

1. make clean
2. make all flash
3. make monitor
4. reset chip

### Code to reproduce this issue

```cpp
#include <stddef.h>
#include <stdio.h>
#include <stdint.h>
#include <string.h>
#include <stdlib.h>

#include "esp_system.h"

#include "driver/uart.h"

uint32_t user_rf_cal_sector_set(void) {
    flash_size_map size_map = system_get_flash_size_map();
    uint32_t rf_cal_sec = 0;

    switch (size_map) {
        case FLASH_SIZE_4M_MAP_256_256:
            rf_cal_sec = 128 - 5;
            break;

        case FLASH_SIZE_8M_MAP_512_512:
            rf_cal_sec = 256 - 5;
            break;

        case FLASH_SIZE_16M_MAP_512_512:
        case FLASH_SIZE_16M_MAP_1024_1024:
            rf_cal_sec = 512 - 5;
            break;

        case FLASH_SIZE_32M_MAP_512_512:
        case FLASH_SIZE_32M_MAP_1024_1024:
            rf_cal_sec = 1024 - 5;
            break;

        case FLASH_SIZE_64M_MAP_1024_1024:
            rf_cal_sec = 2048 - 5;
            break;

        case FLASH_SIZE_128M_MAP_1024_1024:
            rf_cal_sec = 4096 - 5;
            break;

        default:
            rf_cal_sec = 0;
            break;
    }

    return rf_cal_sec;
}

[sdkconfig.txt](https://github.com/espressif/ESP8266_RTOS_SDK/files/2130188/sdkconfig.txt)
void user_init(void) {

	printf("SDK version:%s rom %d\n", system_get_sdk_version(), system_upgrade_userbin_check());
	printf("Time = %d\n", system_get_time());
	printf("Chip id = 0x%x\n", system_get_chip_id());
	printf("CPU freq = %d MHz\n", system_get_cpu_freq());
	printf("Flash size map = %d\n", system_get_flash_size_map());
	printf("Free heap size = %d\n", system_get_free_heap_size());
}

## Debug Logs

```
ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 4544, room 16
0x40100000: _stext at ??:?

tail 0
chksum 0xe2
load 0x3ffe8008, len 3960, room 8
tail 0
chksum 0x05
csum 0x05
I (21) boot: ESP-IDF v2.0.0-492-g305a338-dirty 2nd stage bootloader
I (21) boot: compile time 22:03:32
D (21) boot: magic e9
D (23) boot: segments 02
D (25) boot: spi_mode 00
D (28) boot: spi_speed 00
D (30) boot: spi_size 03
I (33) boot: SPI Speed      : 40MHz
I (37) boot: SPI Mode       : QIO
I (41) boot: SPI Flash Size : 2MB
D (44) boot: mapped partition table 0x8000 at 0x40208000
D (49) flash_parts: partition table verified, 3 entries
I (54) boot: Partition Table:
I (58) boot: ## Label            Usage          Type ST Offset   Length
D (64) boot: load partition table entry 0x3ffffe04
D (69) boot: type=1 subtype=2
I (72) boot:  0 nvs              WiFi data        01 02 00009000 00006000
D (79) boot: load partition table entry 0x3ffffe04
D (83) boot: type=1 subtype=1
I (86) boot:  1 phy_init         RF data          01 01 0000f000 00001000
D (93) boot: load partition table entry 0x3ffffe04
D (98) boot: type=0 subtype=0
I (101) boot:  2 factory          factory app      00 00 00010000 00100000
I (108) boot: End of partition table
D (112) boot: Trying partition index -1 offs 0x10000 size 0x100000
D (118) esp_image: reading image header @ 0x10000
D (122) esp_image: image header: 0xe9 0x04 0x00 0x01 40233a00
I (128) esp_image: segment 0: paddr=0x00010010 vaddr=0x40210010 size=0x3e078 (254072) map
I (185) esp_image: segment 1: paddr=0x0004e090 vaddr=0x3ffe8000 size=0x003f4 (  1012) load
I (185) esp_image: segment 2: paddr=0x0004e48c vaddr=0x3ffe83f4 size=0x0011c (   284) load
I (192) esp_image: segment 3: paddr=0x0004e5b0 vaddr=0x40100000 size=0x06ecc ( 28364) load
0x40100000: _stext at ??:?

I (206) boot: Loaded app from partition at offset 0x10000
      PC: 0x402398c2        PS: 0x00000030        A0: 0x40247f5d        A1: 0x3ffff940
0x402398c2: xQueueReceiveFromISR at C:/Users/Jeroen/Downloads/esp8266/ESP8266_RTOS/msys32/home/Jeroen/esp/ESP8266_RTOS_SDK/components/freertos/freerto
s/queue.c:2332

0x40247f5d: lock_acquire_generic at C:/Users/Jeroen/Downloads/esp8266/ESP8266_RTOS/msys32/home/Jeroen/esp/ESP8266_RTOS_SDK/components/newlib/newlib/po
rt/locks.c:168

      A2: 0xf4e1626b        A3: 0x00000000        A4: 0x3ffff960        A5: 0x00000010
      A6: 0x08084297        A7: 0x60000200        A8: 0x00000020        A9: 0x401017c4
0x401017c4: user_fatal_exception_handler at ??:?

     A10: 0x3ffee474       A11: 0x00000102       A12: 0xf4e1626b       A13: 0x00000000
     A14: 0x3ffff960       A15: 0x00000001       SAR: 0x00000005  EXCCAUSE: 0x00000009
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
[sdkconfig.txt](https://github.com/espressif/ESP8266_RTOS_SDK/files/2130193/sdkconfig.txt)

- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
[my_first_project.elf.zip](https://github.com/espressif/ESP8266_RTOS_SDK/files/2130191/my_first_project.elf.zip)

- [ ] coredump (This provides stacks of tasks.)

cranphin, 2018-06-24:
Oops! It did ask me which log level to use after the update, it seemed a good idea to set it more verbose.. :)
Thanks for finding that out, I'll try a lower log level soon! :)

27-301, 2018-07-10:
@cranphin The log level issue is fixed in the latest master, please check. Thanks.

cranphin, 2018-07-10:
I tried, can't build now, see: https://github.com/espressif/ESP8266_RTOS_SDK/issues/252 :)

cranphin, 2018-07-12:
Latest SDK version still seems to cause a problem.

Using the SDK and example project_template from ESP8266_RTOS_SDK revision 7005cee,
and https://dl.espressif.com/dl/xtensa-lx106-elf-win32-1.22.0-88-gde0bdc1-4.8.5.tar.gz on msys2.

If I set DEBUG logging in menuconfig, I get this bootloop:
```
 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x40100000, len 4744, room 16
0x40100000: _stext at ??:?

tail 8
chksum 0x1a
load 0x3ffe8008, len 4008, room 0
tail 8
chksum 0x84
csum 0x84
I (21) boot: ESP-IDF v2.0.0-616-g7005cee-dirty 2nd stage bootloader
I (22) boot: compile time 21:53:27
D (22) boot: magic e9
D (23) boot: segments 02
D (26) boot: spi_mode 00
D (28) boot: spi_speed 00
D (30) boot: spi_size 03
I (33) boot: SPI Speed      : 40MHz
I (37) boot: SPI Mode       : QIO
I (41) boot: SPI Flash Size : 2MB
D (44) boot: mapped partition table 0x8000 at 0x40208000
D (49) flash_parts: partition table verified, 3 entries
I (54) boot: Partition Table:
I (58) boot: ## Label            Usage          Type ST Offset   Length
D (65) boot: load partition table entry 0x3ffffe04
D (69) boot: type=1 subtype=2
I (72) boot:  0 nvs              WiFi data        01 02 00009000 00006000
D (79) boot: load partition table entry 0x3ffffe04
D (83) boot: type=1 subtype=1
I (87) boot:  1 phy_init         RF data          01 01 0000f000 00001000
D (93) boot: load partition table entry 0x3ffffe04
D (98) boot: type=0 subtype=0
I (101) boot:  2 factory          factory app      00 00 00010000 00100000
I (108) boot: End of partition table
D (112) boot: Trying partition index -1 offs 0x10000 size 0x100000
D (118) esp_image: reading image header @ 0x10000
D (122) esp_image: image header: 0xe9 0x04 0x00 0x01 4022888c
I (128) esp_image: segment 0: paddr=0x00010010 vaddr=0x40210010 size=0x38a20 (231968) map
I (183) esp_image: segment 1: paddr=0x00048a38 vaddr=0x3ffe8000 size=0x00ee0 (  3808) load
I (184) esp_image: segment 2: paddr=0x00049920 vaddr=0x3ffe8ee0 size=0x00110 (   272) load
I (190) esp_image: segment 3: paddr=0x00049a38 vaddr=0x40100000 size=0x075b4 ( 30132) load
0x40100000: _stext at ??:?

I (204) boot: Loaded app from partition at offset 0x10000
Fatal exception (3):
epc1=0x40248824, epc2=0x00000000, epc3=0x00000000, excvaddr=0x402262e8, depc=0x00000000
0x40248824: memchr at /home/dongheng/workspace/espressif/tmp/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-lx106-elf/newli
b/libc/string/../../../.././newlib/libc/string/memchr.c:128

0x402262e8: chip_v6_unset_chanfreq at ??:?

Fatal exception (3):
epc1=0x40248824, epc2=0x00000000, epc3=0x00000000, excvaddr=0x402262e8, depc=0x00000000
0x40248824: memchr at /home/dongheng/workspace/espressif/tmp/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-lx106-elf/newli
b/libc/string/../../../.././newlib/libc/string/memchr.c:128

0x402262e8: chip_v6_unset_chanfreq at ??:?
```

If I set INFO logging in menuconfig, there is no error:
```
ets Jan  8 2013,rst cause:2, boot mode:(3,0)

load 0x40100000, len 4744, room 16
0x40100000: _stext at ??:?

tail 8
chksum 0x1a
load 0x3ffe8008, len 4008, room 0
tail 8
chksum 0x83
csum 0x83
I (21) boot: ESP-IDF v2.0.0-616-g7005cee-dirty 2nd stage bootloader
I (22) boot: compile time 21:55:37
D (22) boot: magic e9
D (23) boot: segments 02
D (26) boot: spi_mode 00
D (28) boot: spi_speed 00
D (30) boot: spi_size 03
I (33) boot: SPI Speed      : 40MHz
I (37) boot: SPI Mode       : QIO
I (41) boot: SPI Flash Size : 2MB
D (44) boot: mapped partition table 0x8000 at 0x40208000
D (49) flash_parts: partition table verified, 3 entries
I (54) boot: Partition Table:
I (58) boot: ## Label            Usage          Type ST Offset   Length
D (65) boot: load partition table entry 0x3ffffe04
D (69) boot: type=1 subtype=2
I (72) boot:  0 nvs              WiFi data        01 02 00009000 00006000
D (79) boot: load partition table entry 0x3ffffe04
D (83) boot: type=1 subtype=1
I (87) boot:  1 phy_init         RF data          01 01 0000f000 00001000
D (93) boot: load partition table entry 0x3ffffe04
D (98) boot: type=0 subtype=0
I (101) boot:  2 factory          factory app      00 00 00010000 00100000
I (108) boot: End of partition table
D (112) boot: Trying partition index -1 offs 0x10000 size 0x100000
D (118) esp_image: reading image header @ 0x10000
D (122) esp_image: image header: 0xe9 0x04 0x00 0x01 402285c0
I (128) esp_image: segment 0: paddr=0x00010010 vaddr=0x40210010 size=0x385e0 (230880) map
I (183) esp_image: segment 1: paddr=0x000485f8 vaddr=0x3ffe8000 size=0x00edc (  3804) load
I (184) esp_image: segment 2: paddr=0x000494dc vaddr=0x3ffe8edc size=0x00110 (   272) load
I (190) esp_image: segment 3: paddr=0x000495f4 vaddr=0x40100000 size=0x074a0 ( 29856) load
0x40100000: _stext at ??:?

I (204) boot: Loaded app from partition at offset 0x10000
SDK version:v2.0.0-616-g7005cee-dirty
```

27-301, 2018-07-13:
@donghengqaz @FayeY Please check the log level.