espressif/ESP8266_RTOS_SDK #238: Some questions about CPU CLOCK and FreeRTOS configCPU_CLOCK_HZ（80MHz or 160MHz）?

ichenxing, 2018-06-24:
## Environment

- Development Kit:        cygwin
- IDF version (``git rev-parse --short HEAD`` to get the commit id.): 
    //305a338
- Development Env:      [Make]
- Operating System:     [Windows]
- Power Supply:            [USB]

## Problem Description

```
if(system_update_cpu_freq(SYS_CPU_160MHZ) == false)
{
    	dbg("\nERROR: system_update_cpu_freq failed.\n");
}
```
修改系统时钟之后，FreeRTOSConfig.h的时钟要不要跟着修改？
`#define configCPU_CLOCK_HZ			( ( unsigned long ) 80000000 )`

官方文档上这样介绍的：
Please do not revise FreeRTOSConfig.h. Revision of the head file can only be
done by libraries in the SDK.

尝试着修改系统时钟按照上述的方法；

### Expected Behavior

```
const portTickType xDelay = pdMS_TO_TICKS(1000); 

for(;;)
{
        LedToggle();
        vTaskDelay( xDelay );
}
```

延时函数控制LED进行1s信号翻转，期望应该是与1s有很小的误差；

### Actual Behavior

信号翻转的实际时间小于1s很多，大概在500ms左右；

请问怎么操作才可以使系统时钟由默认的80MHz改成160MHz？

27-301, 2018-06-24:
hi, it's still under refactor. Do you want to configure 160MHz as default freq?

ichenxing, 2018-06-24:
@27-301 <it's still under refactor> means i can't use the function to modify cpu frequency now, then how can i configure 160MHz as default freq?