espressif/ESP8266_RTOS_SDK #243: Please tell me which SDK release version is stable.

aeris-ming, 2018-06-28:
I really want to focus on APP development.
1.git clone this SDK
2.git check out v2.0.0(It is even version,I think it should be stable)
3.configure and make
4.flash to ESP-12 
5.reboot again and again
6.then use google and baidu,try to find out reboot reason for about 3hours late at night
7.I used to download this SDK and it works,then I try the old sdk,ESP-12 works well
Debug OutPut:OS SDK ver: 2.0-dev(79c549f) compiled @ Mar 12 2018 15:38:05
8.It is version 2.0 too,Why?!!!

27-301, 2018-06-29:
the commit of v2.0.0 is 936c92a7.

can you provide you demo? configuration? flash address of each binary? and also reboot log.

Advise you to use master branch, although it's under testing now, but will update frequently. v2.x will only have bugfix, but no new features include. recent master will be compatible with esp-idf APIs, the application can be shared between esp32 & esp8266 in some cases. you can use http_request for test.

Thanks

aeris-ming, 2018-07-01:
make menuconfig feature not work
commit bc65f24
Merge: 7e1efaa 3f6158a

/ESP8266/usr/dtu/main/fake_eeprom_help.c:13:2: error: unknown type name 'SpiFlashOpResult'
  SpiFlashOpResult result;

aeris-ming, 2018-07-01:
Another issue is It takes to long to compile third party lib

aeris-ming, 2018-07-01:
Too many MACO API and headers changed

/home/aeris/ESP8266/usr/dtu/main/smart_config.c:28:1: error: unknown type name 'esp_udp'
 static esp_udp ssdp_udp;
 ^
/home/aeris/ESP8266/usr/dtu/main/smart_config.c:30:1: error: unknown type name 'os_timer_t'
 static os_timer_t ssdp_time_serv;
 ^
/home/aeris/ESP8266/usr/dtu/main/smart_config.c:37:26: error: 'memset' undeclared here (not in a function)
     (airkiss_memset_fn)& memset,
                          ^
/home/aeris/ESP8266/usr/dtu/main/smart_config.c:38:26: error: 'memcpy' undeclared here (not in a function)
     (airkiss_memcpy_fn)& memcpy,
                          ^
/home/aeris/ESP8266/usr/dtu/main/smart_config.c:39:26: error: 'memcmp' undeclared here (not in a function)
     (airkiss_memcmp_fn)& memcmp,
                          ^
/home/aeris/ESP8266/usr/dtu/main/smart_config.c: In function 'airkiss_wifilan_time_callback':
/home/aeris/ESP8266/usr/dtu/main/smart_config.c:50:9: warning: implicit declaration of function 'os_timer_disarm' [-Wimplicit-function-declaration]
         os_timer_disarm(&ssdp_time_serv);//s
         ^
/home/aeris/ESP8266/usr/dtu/main/smart_config.c:54:13: error: request for member 'remote_port' in something not a structure or union
     ssdp_udp.remote_port = DEFAULT_LAN_PORT;
             ^
/home/aeris/ESP8266/usr/dtu/main/smart_config.c:55:13: error: request for member 'remote_ip' in something not a structure or union
     ssdp_udp.remote_ip[0] = 255;
             ^
/home/aeris/ESP8266/usr/dtu/main/smart_config.c:56:13: error: request for member 'remote_ip' in something not a structure or union
     ssdp_udp.remote_ip[1] = 255;
             ^
/home/aeris/ESP8266/usr/dtu/main/smart_config.c:57:13: error: request for member 'remote_ip' in something not a structure or union
     ssdp_udp.remote_ip[2] = 255;

aeris-ming, 2018-07-01:
v2.0 936c92a not work too,Device reboot again and again

[log.txt](https://github.com/espressif/ESP8266_RTOS_SDK/files/2152367/log.txt)

27-301, 2018-07-02:
1. you need to export IDF_PATH
2. the first time compile will need a lot of time
3. yes, smartconfig demo needs to be fixed
4. driver is still under refactoring

closed 2018-07-09 by aeris-ming