espressif/ESP8266_RTOS_SDK #244: 自己编译第三方lwip源代码，运行提示STATION_WRONG_PASSWORD类错误。

xiangjiangnan, 2018-06-30:
项目开发中，由于程序运行中出现了如下问题：
"tiT"(stack_size = 4,task handle = 40107a00) overflow the heap_size.
"tiT"(stack_size = 4,task handle = 40107a00) overflow the heap_size.
"tiT"(stack_size = 4,task handle = 40107a00) overflow the heap_size.
"tiT"(stack_size = 4,task handle = 40107a00) overflow the heap_size.
我尝试编译lwip源代码获取一些lwip task的堆栈使用情况信息，所以自己编译了third_party中的源代码，生成的库在运行中，WiFi始终连不上，我使用此函数wifiStatus = wifi_station_get_connect_status();查看问题，显示STATION_WRONG_PASSWORD类错误，可是我的password确信是正确的，因为只要使用lib文件夹下的liblwip.a是可以正常连接的。
请问以上问题原因是什么？该如何解决？谢谢！

xiangjiangnan, 2018-07-01:
已找到问题，使用的eclipse编译环境存在问题，换成virtual box 编译环境就好了。

27-301, 2018-07-02:
能具体些吗？是因为 toolchain 的版本不一样？或是其他。这样对其他人也会有更大的帮助。谢谢

xiangjiangnan, 2018-07-11:
我采用的是安信可公司提供的eclipse编译环境，没有修改任何配置，单独把第三方提供lwip源代码用来编译，编译成功，但是一直连不上WiFi，后来换用了官方的virtualbox环境，编译生成lib库，放进工程是用，就正常了。具体问题在哪里尚不清楚。
谢谢。

27-301, 2018-07-12:
OK，有问题再反馈，谢谢。

closed 2018-07-12 by 27-301