espressif/ESP8266_RTOS_SDK #245: Build Error of the newest ESP8266_RTOS_SDK

ichenxing, 2018-07-01:
## Environment

- Development Kit:
- IDF version (``git rev-parse --short HEAD`` to get the commit id.): 
    bc65f24
- Development Env:      [Make]
- Operating System:     [Ubuntu]
- Power Supply:         [USB]

## Problem Description

//Detailed problem description goes here.
esp8266@esp8266-VirtualBox:~/Share/esp$ cd http_request/
esp8266@esp8266-VirtualBox:~/Share/esp/http_request$ make all
WARNING: Failed to find Xtensa toolchain, may need to alter PATH or set one in the configuration menu
WARNING: Failed to find Xtensa toolchain, may need to alter PATH or set one in the configuration menu
LD /mnt/Share/esp/http_request/build/http-request.elf
/opt/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: /mnt/Share/esp/http_request/build/http-request.elf section `.text' will not fit in region `iram1_0_seg'
/opt/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/lib/libstdc++.a(cp-demangle.o): In function `d_append_string':
/home/matt/ct/crosstool-NG/.build/xtensa-lx106-elf/build/build-cc-final/xtensa-lx106-elf/libstdc++-v3/libsupc++/cp-demangle.c:3732:(.text+0x3c5): dangerous relocation: call0: call target out of range: strlen
/opt/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/lib/libstdc++.a(cp-demangle.o): In function `d_append_num':
/home/matt/ct/crosstool-NG/.build/xtensa-lx106-elf/build/build-cc-final/xtensa-lx106-elf/libstdc++-v3/libsupc++/cp-demangle.c:3739:(.text+0x3ee): dangerous relocation: call0: call target out of range: sprintf
/opt/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/lib/libstdc++.a(cp-demangle.o): In function `d_identifier':
/home/matt/ct/crosstool-NG/.build/xtensa-lx106-elf/build/build-cc-final/xtensa-lx106-elf/libstdc++-v3/libsupc++/cp-demangle.c:1617:(.text+0x56e): dangerous relocation: call0: call target out of range: memcmp
/opt/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/lib/libstdc++.a(cp-demangle.o): In function `d_growable_string_resize':
/home/matt/ct/crosstool-NG/.build/xtensa-lx106-elf/build/build-cc-final/xtensa-lx106-elf/libstdc++-v3/libsupc++/cp-demangle.c:3620:(.text+0x74a): dangerous relocation: call0: call target out of range: realloc
/home/matt/ct/crosstool-NG/.build/xtensa-lx106-elf/build/build-cc-final/xtensa-lx106-elf/libstdc++-v3/libsupc++/cp-demangle.c:3623:(.text+0x753): dangerous relocation: call0: call target out of range: free
/opt/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/lib/libstdc++.a(cp-demangle.o): In function `d_growable_string_append_buffer':
/home/matt/ct/crosstool-NG/.build/xtensa-lx106-elf/build/build-cc-final/xtensa-lx106-elf/libstdc++-v3/libsupc++/cp-demangle.c:3649:(.text+0x7aa): dangerous relocation: call0: call target out of range: memcpy
/opt/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/lib/libstdc++.a(cp-demangle.o): In function `d_print_comp':
/home/matt/ct/crosstool-NG/.build/xtensa-lx106-elf/build/build-cc-final/xtensa-lx106-elf/libstdc++-v3/libsupc++/cp-demangle.c:4113:(.text+0xa33): dangerous relocation: call0: call target out of range: strncmp
/home/matt/ct/crosstool-NG/.build/xtensa-lx106-elf/build/build-cc-final/xtensa-lx106-elf/libstdc++-v3/libsupc++/cp-demangle.c:4598:(.text+0xeb1): dangerous relocation: call0: call target out of range: strcmp
/home/matt/ct/crosstool-NG/.build/xtensa-lx106-elf/build/build-cc-final/xtensa-lx106-elf/libstdc++-v3/libsupc++/cp-demangle.c:4625:(.text+0xf31): dangerous relocation: call0: call target out of range: strcmp
/home/matt/ct/crosstool-NG/.build/xtensa-lx106-elf/build/build-cc-final/xtensa-lx106-elf/libstdc++-v3/libsupc++/cp-demangle.c:4626:(.text+0xf45): dangerous relocation: call0: call target out of range: strcmp
/home/matt/ct/crosstool-NG/.build/xtensa-lx106-elf/build/build-cc-final/xtensa-lx106-elf/libstdc++-v3/libsupc++/cp-demangle.c:4666:(.text+0xfd1): dangerous relocation: call0: call target out of range: strcmp
/home/matt/ct/crosstool-NG/.build/xtensa-lx106-elf/build/build-cc-final/xtensa-lx106-elf/libstdc++-v3/libsupc++/cp-demangle.c:4680:(.text+0x1012): dangerous relocation: call0: call target out of range: strcmp
/home/matt/ct/crosstool-NG/.build/xtensa-lx106-elf/build/build-cc-final/xtensa-lx106-elf/libstdc++-v3/libsupc++/cp-demangle.c:4686:(.text+0x103e): dangerous relocation: call0: call target out of range: strcmp
/home/matt/ct/crosstool-NG/.build/xtensa-lx106-elf/build/build-cc-final/xtensa-lx106-elf/libstdc++-v3/libsupc++/cp-demangle.c:4711:(.text+0x10ad): dangerous relocation: call0: call target out of range: strcmp
/opt/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/lib/libstdc++.a(cp-demangle.o): In function `d_unqualified_name':
/home/matt/ct/crosstool-NG/.build/xtensa-lx106-elf/build/build-cc-final/xtensa-lx106-elf/libstdc++-v3/libsupc++/cp-demangle.c:1477:(.text+0x288a): dangerous relocation: call0: call target out of range: strcmp
/opt/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/lib/libstdc++.a(cp-demangle.o): In function `d_expression':
/home/matt/ct/crosstool-NG/.build/xtensa-lx106-elf/build/build-cc-final/xtensa-lx106-elf/libstdc++-v3/libsupc++/cp-demangle.c:3047:(.text+0x2c13): dangerous relocation: call0: call target out of range: strcmp
/home/matt/ct/crosstool-NG/.build/xtensa-lx106-elf/build/build-cc-final/xtensa-lx106-elf/libstdc++-v3/libsupc++/cp-demangle.c:3106:(.text+0x2cf8): dangerous relocation: call0: call target out of range: strcmp
/home/matt/ct/crosstool-NG/.build/xtensa-lx106-elf/build/build-cc-final/xtensa-lx106-elf/libstdc++-v3/libsupc++/cp-demangle.c:3108:(.text+0x2d0d): dangerous relocation: call0: call target out of range: strcmp
/home/matt/ct/crosstool-NG/.build/xtensa-lx106-elf/build/build-cc-final/xtensa-lx106-elf/libstdc++-v3/libsupc++/cp-demangle.c:3114:(.text+0x2d17): dangerous relocation: call0: call target out of range: strcmp
/home/matt/ct/crosstool-NG/.build/xtensa-lx106-elf/build/build-cc-final/xtensa-lx106-elf/libstdc++-v3/libsupc++/cp-demangle.c:3133:(.text+0x2d63): dangerous relocation: call0: call target out of range: strcmp
/opt/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/lib/libstdc++.a(cp-demangle.o): In function `d_exprlist':
/home/matt/ct/crosstool-NG/.build/xtensa-lx106-elf/build/build-cc-final/xtensa-lx106-elf/libstdc++-v3/libsupc++/cp-demangle.c:2943:(.text+0x2ecd): dangerous relocation: call0: call target out of range: strncmp
/opt/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/lib/libstdc++.a(cp-demangle.o): In function `d_demangle_callback':
/home/matt/ct/crosstool-NG/.build/xtensa-lx106-elf/build/build-cc-final/xtensa-lx106-elf/libstdc++-v3/libsupc++/cp-demangle.c:5396:(.text+0x2f1b): dangerous relocation: call0: call target out of range: strlen
/opt/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/lib/libstdc++.a(cp-demangle.o): In function `d_mangled_name':
/home/matt/ct/crosstool-NG/.build/xtensa-lx106-elf/build/build-cc-final/xtensa-lx106-elf/libstdc++-v3/libsupc++/cp-demangle.c:1117:(.text+0x3015): dangerous relocation: call0: call target out of range: strlen
/opt/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/lib/libstdc++.a(cp-demangle.o): In function `d_demangle_callback':
/home/matt/ct/crosstool-NG/.build/xtensa-lx106-elf/build/build-cc-final/xtensa-lx106-elf/libstdc++-v3/libsupc++/cp-demangle.c:5433:(.text+0x304e): dangerous relocation: call0: call target out of range: strlen
/home/matt/ct/crosstool-NG/.build/xtensa-lx106-elf/build/build-cc-final/xtensa-lx106-elf/libstdc++-v3/libsupc++/cp-demangle.c:5460:(.text+0x30f6): dangerous relocation: call0: call target out of range: free
/opt/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/lib/libstdc++.a(cp-demangle.o): In function `__cxa_demangle':
/home/matt/ct/crosstool-NG/.build/xtensa-lx106-elf/build/build-cc-final/xtensa-lx106-elf/libstdc++-v3/libsupc++/cp-demangle.c:5530:(.text+0x3121): dangerous relocation: call0: call target out of range: strlen
/home/matt/ct/crosstool-NG/.build/xtensa-lx106-elf/build/build-cc-final/xtensa-lx106-elf/libstdc++-v3/libsupc++/cp-demangle.c:5534:(.text+0x312f): dangerous relocation: call0: call target out of range: strcpy
[... cut ...]
/opt/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/lib/libstdc++.a(eh_catch.o): In function `__cxa_end_catch':
/home/matt/ct/crosstool-NG/.build/src/gcc-4.8.2/libstdc++-v3/libsupc++/eh_catch.cc:125:(.text.__cxa_end_catch+0x49): dangerous relocation: call0: call target out of range: _Unwind_DeleteException
collect2: error: ld returned 1 exit status
make: *** [/mnt/Share/esp/http_request/build/http-request.elf] Error 1

### Expected Behavior

### Actual Behavior

27-301, 2018-07-02:
hi, please update the toolchain to v4.8.5.

ichenxing, 2018-07-02:
@27-301 please tell me the link to download v4.8.5 toolchain, thanks!

27-301, 2018-07-12:
@ichenxing You can get the new toolchains from here:
Windows: https://dl.espressif.com/dl/xtensa-lx106-elf-win32-1.22.0-88-gde0bdc1-4.8.5.tar.gz
Linux32: https://dl.espressif.com/dl/xtensa-lx106-elf-linux32-1.22.0-88-gde0bdc1-4.8.5.tar.gz
Linux64: https://dl.espressif.com/dl/xtensa-lx106-elf-linux64-1.22.0-88-gde0bdc1-4.8.5.tar.gz
MacOS: https://dl.espressif.com/dl/xtensa-lx106-elf-osx-1.22.0-88-gde0bdc1-4.8.5.tar.gz

It's based on https://github.com/espressif/crosstool-NG.git

27-301, 2018-07-18:
@ichenxing @THKDev Has updated url in README in commit a29928de.

closed 2018-07-18 by 27-301