espressif/ESP8266_RTOS_SDK #247: Non OTA vesion works But OTA version crash or reboot again and agin

aeris-ming, 2018-07-01:
Please read issue 246
https://github.com/espressif/ESP8266_RTOS_SDK/issues/246

I closed it unexpected.

aeris-ming, 2018-07-01:
https://github.com/espressif/ESP8266_RTOS_SDK/blob/v2.0.0/ld/eagle.app.v6.ld
I think the parameter address is wrong.Please check it.

aeris-ming, 2018-07-01:
Very interesting,I think this device it 512KByte or Parameter address is wrong.
It works when flash system parameter to 0x7C000.
I think an ESP8266 with 512KB do not have enough room for OTA.
[image]
[image]
[image]

27-301, 2018-07-02:
hi, for 512KB flash, if the binary(user.bin) is larger than 256KB-4KB*6, it can not support OTA. The 6 sectors reversed as 3 system parameter sectors, 1 bootloader, 1 init data bin sector, 1 cal data sector. For eagle.app.v6.ld, it's only used for non OTA mode.

aeris-ming, 2018-07-10:
Change spi flash configure to DIO fix this issue

closed 2018-07-10 by aeris-ming