espressif/ESP8266_RTOS_SDK #251: esp8266内存对齐bug。

xiangjiangnan, 2018-07-07:
最近在开发 esp8266程序的时候，发现esp8266存在内存对齐的问题，明显的表现在动态分配内存的时候，于是写了如下测试代码：
typedef struct{
	int32_t i1;
	uint8_t data[];
} test_t;

void func(){
	test_t * s1 = (test_t*) malloc(sizeof(test_t)*3);
	test_t * s2 = (test_t*) (s1->data+1);
	s2->i1 = 11;
	os_printf("s2->i1 = %d\n", s2->i1);
	free(s1);
}
esp8266程序会不断重启，并抛出exception 9的错误，针对这种问题，ESP8266 SDK中，对硬件unalign 内存的硬件异常进行处理，一旦出现后，按常理来说应该利用l8i指令修复不对齐的问题，而不是直接当成重大错误处理。请问是不是该从汇编的角度对SDK进行这方面优化？

27-301, 2018-07-09:
这是 xtensa CPU 的特性，不允许对非四字节对齐的 int32_t 型进行读写操作。你这里 s2 的地址是强制类型转换得到的， s2->i1 的地址为非四字节对齐，因此会造成 exception 9 的异常。

27-301, 2018-07-10:
@xiangjiangnan any response about this issue?

xiangjiangnan, 2018-07-11:
我的开发场景是这样的，我每次设备启动时需要从flash中获取数据，为了节省内存，我采用动态内存方式获取数据，在对数据进行强转的过程中，由于exception 9异常，致使整个程序无法正常运行，所以我的建议是不要把这种问题简单的作为一个异常抛出，导致整的程序崩溃，其他的嵌入式设备就不存在这种情况。
谢谢。

27-301, 2018-07-12:
了解，需要再调查一下。

BalrogMithrandir, 2018-07-13:
您可以看一下c专家编程里的总线错误那一小节，个人认为这个是c的特性，对这种动态申请的内存，还是按字节来读取比较安全，按照结构体或者其类型对指针强制转换，还是不安全的（虽然大多数cpu和编译器替我们规避了这个问题）。

xiangjiangnan, 2018-07-14:
好的，谢谢。

closed 2018-09-18 by FayeY