espressif/ESP8266_RTOS_SDK #252: Current version won't build example project_template or any other project

cranphin, 2018-07-10:
## Environment

- Development Kit: esp12e + white adapter
- IDF version (``git rev-parse --short HEAD`` to get the commit id.): 
    cb0c5ee5fce091b03a2af025f4bc59e8c188d0f0
- Development Env:      Make
- Operating System:     Windows/msys2
- Power Supply:         USB

## Problem Description

After updating to the latest git version (git status shows no local changes),
and copying the current example 'project_template'.
I can no longer build, instead I get errors:

```
LD build/project_template.elf
c:/users/jeroen/downloads/esp8266/esp8266_rtos/msys32/opt/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: C:/Users/Jeroen/Downloads/esp8266/ESP8266_RTOS/msys32/home/Jeroen/project_template/build/project_template.elf section `.text' will not fit in region `iram1_0_seg'
c:/users/jeroen/downloads/esp8266/esp8266_rtos/msys32/opt/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/lib\libstdc++.a(cp-demangle.o): In function `d_identifier':
d:\ivan\projects\arduinoesp\toolchain\dl\gcc-xtensa\build-2\xtensa-lx106-elf\libstdc++-v3\libsupc++/cp-demangle.c:1617:(.text+0x5ce): dangerous relocation: call0: call target out of range: memcmp
c:/users/jeroen/downloads/esp8266/esp8266_rtos/msys32/opt/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/lib\libstdc++.a(cp-demangle.o): In function `d_growable_string_resize':
d:\ivan\projects\arduinoesp\toolchain\dl\gcc-xtensa\build-2\xtensa-lx106-elf\libstdc++-v3\libsupc++/cp-demangle.c:3620:(.text+0x77e): dangerous relocation: call0: call target out of range: realloc
.
.
Many more dangerous relocation
```

### Expected Behavior

Building works

### Actual Behavior

I get the mentioned errors

### Steps to repropduce

1. Update to latest git version
2. copy/paste project_template from examples folder
3. make all, choose defaults for all menu items

cranphin, 2018-07-10:
I guess this issue might be similar to https://github.com/espressif/ESP8266_RTOS_SDK/issues/245 ?

`$ /opt/xtensa-lx106-elf/xtensa-lx106-elf/bin/c++ --version`
c++.exe (GCC) **4.8.2**

I guess I also need 4.8.5? Where to find the windows build for that?
The ESP8266_RTOS_SDK Readme.md points to the old version (from esp8266 Arduino core, 2.3.0 release).

luca-88-, 2018-07-10:
Have the same issue, but I've build xtensa toolchain version 4.8.5 with crosstool-ng and work now !

cranphin, 2018-07-10:
@elettropic where do you get the source for it, and how does building work? :)

redchenjs, 2018-07-10:
@cranphin https://github.com/esp8266/esp8266-wiki/wiki/Toolchain

cranphin, 2018-07-11:
Nice! Trying to build on my msys2 now (the one I use for this sdk), but running into the dreaded:
"zconf.tab.o:zconf.tab.c:(.text+0x167b): undefined reference to `libintl_gettext'"

Which OS/env did you build on? And which OS/env do you use the SDK on? :)

cranphin, 2018-07-11:
Never mind, that bit is fixed with:

```diff
diff --git a/kconfig/Makefile b/kconfig/Makefile
index 3474e5c6..c5349243 100644
--- a/kconfig/Makefile
+++ b/kconfig/Makefile
@@ -35,20 +35,21 @@ conf_SRC = conf.c
 conf_OBJ = $(patsubst %.c,%.o,$(conf_SRC))
 conf_DEP = $(patsubst %.o,%.dep,$(conf_OBJ))
 $(conf_OBJ) $(conf_DEP): CFLAGS += $(INTL_CFLAGS)
+conf: LDFLAGS += -lintl

 # What's needed to build 'mconf'
 mconf_SRC = mconf.c
 mconf_OBJ = $(patsubst %.c,%.o,$(mconf_SRC))
 mconf_DEP = $(patsubst %.c,%.dep,$(mconf_SRC))
 $(mconf_OBJ) $(mconf_DEP): CFLAGS += $(NCURSES_CFLAGS) $(INTL_CFLAGS)
-mconf: LDFLAGS += $(NCURSES_LDFLAGS)
+mconf: LDFLAGS += -lintl $(NCURSES_LDFLAGS)

 # What's needed to build 'nconf'
 nconf_SRC = nconf.c nconf.gui.c
 nconf_OBJ = $(patsubst %.c,%.o,$(nconf_SRC))
 nconf_DEP = $(patsubst %.c,%.dep,$(nconf_SRC))
 $(nconf_OBJ) $(nconf_DEP): CFLAGS += $(INTL_CFLAGS) -I/usr/include/ncurses
-nconf: LDFLAGS += -lmenu -lpanel -lncurses
+nconf: LDFLAGS += -lintl -lmenu -lpanel -lncurses

 # Under Cygwin, we need to auto-import some libs (which ones, exactly?)
 # for mconf and nconf to lin properly.
```

cranphin, 2018-07-11:
Right, or I use the 'xtensa-1.22.x' branch, so:
git clone -b xtensa-1.22.x git://github.com/jcmvbkbc/crosstool-NG.git

That also fixes "undefined reference to `libintl_gettext'", but the case insensitive issue I still can't solve.

Supposedly you can change your windows registry, but I'm affraid of breaking other windows programs.. :)

elettropic, 2018-07-11:
I've succesfully compiled toolchain under linux (ubuntu 18.04) with crosstool-ng from this link https://github.com/espressif/crosstool-NG.git

cranphin, 2018-07-11:
@igrr Hi! It looks like https://github.com/espressif/crosstool-NG.git is set up to build esp32 cross to windows toolchains, could it also build a lx106/esp8266? :) I'm finding it very hard to build in msys2 ;)

27-301, 2018-07-12:
@cranphin I'm trying to use CI to build toolchains for esp8266, but now has some issues. If possible, i can provide some test version for your test. Thanks

27-301, 2018-07-12:
@cranphin Can you help to test the toolchain for windows?
https://dl.espressif.com/dl/xtensa-lx106-elf-win32-1.22.0-88-gde0bdc1-4.8.5.tar.gz

it based on https://github.com/espressif/crosstool-NG.git

27-301, 2018-07-12:
@cranphin Can you help to test the toolchain generated by CI?
https://dl.espressif.com/dl/xtensa-lx106-elf-win32-1.22.0-88-gde0bdc1-4.8.5.tar.gz

It's based on https://github.com/espressif/crosstool-NG.git

For other platforms:
Linux32: https://dl.espressif.com/dl/xtensa-lx106-elf-linux32-1.22.0-88-gde0bdc1-4.8.5.tar.gz
Linux64: https://dl.espressif.com/dl/xtensa-lx106-elf-linux64-1.22.0-88-gde0bdc1-4.8.5.tar.gz
MacOS: https://dl.espressif.com/dl/xtensa-lx106-elf-osx-1.22.0-88-gde0bdc1-4.8.5.tar.gz

cranphin, 2018-07-12:
Sweet! :)

The example project_template  from ESP8266_RTOS_SDK revision 7005cee compiles and flashes with https://dl.espressif.com/dl/xtensa-lx106-elf-win32-1.22.0-88-gde0bdc1-4.8.5.tar.gz on msys2.
Of course it's an empty project, so doesn't do much, but it's enough to close this issue :)

closed 2018-07-12 by cranphin