espressif/ESP8266_RTOS_SDK #253: State of current API (eg. deep sleep)

THKDev, 2018-07-11:
How close is the current state of master branch API to the [ESP IDF Documentation](https://esp-idf.readthedocs.io/en/latest/api-reference/index.html) ?
Where is the deep sleep API gone? Can not find  `system_deep_sleep` nor `esp_deep_sleep_start`.
What is the replacement of `system_get_vdd33` ?

27-301, 2018-07-13:
@Junhao-Espressif Please add esp_sleep.h.

Junhao-Espressif, 2018-08-27:
Hi @THKDev 

We have added support for these interfaces in the v3.0.

Please try.

closed 2018-10-22 by FayeY

musicoliv, 2019-03-29:
Hi @Junhao-Espressif ,
`esp_deep_sleep` has been added, but with a big difference of signature: it handles only uint32 (about 71 minutes of deep sleep). IDF documentation has `void esp_deep_sleep(uint64_t time_in_us)` . As it is already available on NONOS_SDK since release 2.1.0(116b762), this is not a limitation of esp8266 capabilities. Is it possible to update it on RTOS_SDK too?