espressif/ESP8266_RTOS_SDK #262: 关于 MQTT订阅主题之后，不能在回调函数里面收到推送的消息？

xuhongv, 2018-07-23:
- 您好？我这边使用的是 `v2.0 SDK`的分支代码，我发现在跑通你们的`MQTT`的代码之后，我订阅主题成功之后，不做任何事情，服务器下发这个主题的消息，但是在设备端`esp8266`注册的回调方法里面是收不到信息的。

- 我在服务器看到当前客户端的确是向服务器订阅了这个主题；

- 奇怪的是，当我在订阅主题之后，每秒轮训向服务器发生消息时候，这个回调函数才会回调服务下发的消息？？ 这是为什么呢？

xiongyumail, 2018-07-23:
Hi @xuhongv ,
据您的描述，和MQTT服务器连接是没有问题的，但是其它的信息还不能够定位问题。
能否提供关于您的代码流程、Log、哪一个回调函数等信息嘛？

closed 2018-07-23 by xuhongv

ichenxing, 2018-11-27:
大概是因为程序没有执行这段代码
`
/** MQTT start background thread for a client.  After this, MQTTYield should not be called.
*  @param client - the client object to use
*  @return success code
*/
#if defined(MQTT_TASK)
		if ((rc = MQTTStartTask(&client)) != pdPASS) {
			printf("Return code from start tasks is %d\n", rc);
		} else {
			printf("Use MQTTStartTask\n");
		}
#endif
`