espressif/ESP8266_RTOS_SDK #267: 请问下这个最新版esp8266的rtos版本的idf框架，怎么设置下载固件的位置，怎么设置是否支持 OTA ？

xuhongv, 2018-07-27:
- 请问下这个最新版esp8266的rtos版本的idf框架，怎么设置下载固件的位置，怎么设置是否支持 OTA ？

- 我在你们的```project_templat```工程的输入命令```make menuconfig```貌似没有找到是否修改OTA的选型？请问我应该怎么修改？我想入门贵方新出的框架开发8266。

ichenxing, 2018-07-27:
@xuhongv 我找到一些信息关于esp2 OTA的，不知道能不能给到你帮助；
[https://github.com/espressif/esp-idf/blob/master/docs/en/api-guides/partition-tables.rst](url)

xuhongv, 2018-07-27:
@ichenxing 您提供的链接不存在。。不过还是多谢。

ustccw, 2018-07-27:
@xuhongv @ichenxing OTA 功能已完成，在内部测试中，OTA 配置和 ESP-IDF 框架下 OTA 配置类似。

ichenxing, 2018-07-27:
@xuhongv 不要直接点进去，复制到浏览器打开就行

xuhongv, 2018-07-27:
@ustccw 感谢。希望早点完善下这个框架，以及使用说明；毕竟我们都希望看到一个成熟稳定开发框架来开发贵司的芯片。

xuhongv, 2018-07-27:
@ichenxing 好的，感谢您。

27-301, 2018-07-27:
看下这个 https://esp-idf.readthedocs.io/en/latest/api-guides/partition-tables.html，以及 8266 SDK/components/partition_table/ 下的 csv，当前 demo 用的是 singleapp.csv 的。

xuhongv, 2018-07-27:
@27-301 感谢，我先认真阅读下。希望你们也可以尽快出个 ota 的demo。感谢！！

27-301, 2018-08-24:
@xuhongv OTA is supported now, please try the master branch, thanks.

closed 2018-08-24 by 27-301