espressif/ESP8266_RTOS_SDK #269: esp8266和路由器断开连接

BalrogMithrandir, 2018-07-27:
1. .基于乐鑫官网发布的**rtos 2.0**版本sdk进行开发，连接上路由器后，过一会儿8266总是会和路由器断开连接，然后马上有能重新连接成功。之后过一会儿又断开，如此反复，但是间隔不定。
2. .在断开连接时，用api ```wifi_station_get_rssi```查看当前的**rssi**，值在**-60**左右，应该还可以。
3. .断开连接时，sdk会有如下输出。虽然log显示8266扫描不到路由器（ssid为“helloworld”），但是实际上路由器一直在，而且我用其他的wifi模块（运行的固件不是基于2.0sdk的版本）连接，并没有出现该现象。我也试过更换其他的8266wifi模块，但是现象不变。
4. .**请问**这个确实只是因为8266搜索不到路由器的信号，还是有其他原因？有没有可能是软件的原因？谢谢！
```c
[21:03:05.241]收←◆bcn_timout,ap_probe_send_start
[21:03:07.744]收←◆ap_probe_send_over, rest wifi status to disassoc
state: 5 -> 0 (1)
rm 0
pm close 7 0 0/160331701
disconnect from ssid helloworld, reason 200
[21:03:08.747]收←◆reconnect
[21:03:08.868]收←◆scandone
no helloworld found, reconnect after 1s
disconnect from ssid helloworld, reason 201
[Ac.Info][Tbc]Platform_ReleaseSource
reconnect

[21:03:11.403]收←◆scandone
no helloworld found, reconnect after 1s
disconnect from ssid helloworld, reason 201
[Ac.Info][Tbc]Platform_ReleaseSource
reconnect

[21:03:11.626]收←◆scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 2
pm open phy_2,type:2 0 0
cnt 

[21:03:12.159]收←◆
connected with helloworld, channel 1
dhcp client start...
connect to ssid helloworld, channel 1

[21:03:18.924]收←◆ip:10.0.2.13,mask:255.255.255.0,gw:10.0.2.2
[Ac.Info][Tbc]BcInit ok
[Ac.Info][Tbc]LanUdpInit ok
[Ac.Info][Tbc]NtpInit ok

[21:03:19.120]收←◆[Ac.Info][Uart]AC_UartProcess msg code is 7
[Ac.Info][Pro]AC_SendDeviceRegsiter device id is 18FE34D3C1F0
[Ac.Info][Pro]ZC_RecvDataFromMoudle msgcode is 7
[Ac.Info][PltFm]STA mode
[Ac.Info][PltFm]STA mode
[Ac.Info][Pro]Connect gateway
```

ustccw, 2018-07-31:
@BalrogMithrandir 你好，可否提供下面以便我们更好复现问题。
1. 整个测试源码
2. 测试步骤
3. 测试环境
  如路由器配置，开发板型号，无线环境等

ustccw, 2018-08-21:
@BalrogMithrandir 可否试一下附件中的 libphy.a, 测试看看。
[libphy.a.zip](https://github.com/espressif/ESP8266_RTOS_SDK/files/2306360/libphy.a.zip)

closed 2018-10-22 by FayeY