espressif/ESP8266_RTOS_SDK #271: smartconfig crash,v2.0

aeris-ming, 2018-08-01:
1.esp8266 is successfully connected to router
2.call wifi disconnect api and disable reconnect
3.start smartconfig task
4.serial port output "sandone sane error state-->1 sane error state-->4..."
5.then reboot

xuhongv, 2018-08-04:
this is a problem ,  when the Esp8266 is successfully connected to router , then I start smartconfig task again. But is not successfully . 

Now I creat new way to solve the problem , restart the system to into smartconfig task .

Hope the can help you.

aeris-ming, 2018-08-05:
@xuhongv How to fingure out if smartconfig should be start after reset?

xuhongv, 2018-08-06:
when you into smartconfig , set flag number and save on flash . then restart system and read the flag ,if is the smartconfig  flag , we current into smartconfig  mode ,how?

Junhao-Espressif, 2018-09-10:
Hi @aeris-ming 

Could you give me your example?

Thanks!

closed 2018-12-20 by FayeY