espressif/ESP8266_RTOS_SDK #272: Task Stack Problem？

ichenxing, 2018-08-01:
## Environment

- Development Kit: ESP8266-S1
- IDF version (``git rev-parse --short HEAD`` to get the commit id.): 
    bbdb773
- Development Env:      [Make]
- Operating System:     [Ubuntu]
- Power Supply:            [Battery]

## Problem Description
连接上路由器之后，通过tcp连接服务器，出现任务堆栈问题，然后一直重启
`int connect_http_server(const char * http_server, const int http_port)
{
    const struct addrinfo hints = {
        .ai_family = AF_INET,
        .ai_socktype = SOCK_STREAM,
    };
    struct addrinfo *res;
    struct in_addr *addr;
    int s,err;

    while(1) {
	    /* Wait for the callback to set the CONNECTED_BIT in the
	       event group.
	    */
	    xEventGroupWaitBits(wifi_event_group, CONNECTED_BIT,
	                        false, true, portMAX_DELAY);
	    ESP_LOGI(TAG, "Connected to AP");

		if(http_port == 8080){
	    	err = getaddrinfo(http_server, "8080", &hints, &res);
		}else{
	    	err = getaddrinfo(http_server, "80", &hints, &res);
		}

	    if(err != 0 || res == NULL) {
	        ESP_LOGE(TAG, "DNS lookup failed err=%d res=%p", err, res);
	        vTaskDelay(1000 / portTICK_PERIOD_MS);
	        continue;
	    }

	    /* Code to print the resolved IP.
	       Note: inet_ntoa is non-reentrant, look at ipaddr_ntoa_r for "real" code */
	    addr = &((struct sockaddr_in *)res->ai_addr)->sin_addr;
	    ESP_LOGI(TAG, "DNS lookup succeeded. IP=%s", inet_ntoa(*addr));

	    s = socket(res->ai_family, res->ai_socktype, 0);
	    if(s < 0) {
	        ESP_LOGE(TAG, "... Failed to allocate socket.");
	        freeaddrinfo(res);
	        vTaskDelay(1000 / portTICK_PERIOD_MS);
	        continue;
	    }
	    ESP_LOGI(TAG, "... allocated socket");

	    if(connect(s, res->ai_addr, res->ai_addrlen) != 0) {
	        ESP_LOGE(TAG, "... socket connect failed errno=%d", errno);
	        close(s);
	        freeaddrinfo(res);
	        vTaskDelay(4000 / portTICK_PERIOD_MS);
	        continue;
	    }

	    ESP_LOGI(TAG, "... connected");
	    freeaddrinfo(res);

		return s;
    }

}`
[... 115 lines cut ...]
3fff2e40  0000dd40 00000020 40108124 40101440 00000000 ffffffff 00000053 4024eade aaaaaaaa 00000020 00000035 40108440 00000013 3ffef3c8 40107dac 4024d158 
3fff2e80  00000000 00000038 00000037 40100583 00000000 00000000 3fff1b60 4023971e 0000000a 00000000 40107dac 4024d9b0 0000000a 3fff2ef0 40107dac 4024c8e5 
3fff2ec0  4024beb4 3ffef3c8 40107dac 3ffea3a4 00000000 3fff1b60 4022ab5c 4022ab7f 3ffef3c8 0000000b 00000000 40104a45 3ffea104 3ffe8ab0 40108124 402215b1 
3fff2f00  4022499c 00000000 3ffef3c8 4024d9b0 402249af 00000000 3ffef3c8 00000000 3ffe9604 3ffea238 3ffea954 402228b8 3ffe9620 00000013 40302078 4022b8c1 
3fff2f40  ff174104 00000000 3fff1a00 3ffe9d24 3ffee1d8 00000001 3ffe9604 4022fac2 3ffee1d8 3ffe78dc 40107f78 4021d55a 423a7d80 00000402 00000000 4023944e 
3fff2f80  000000b0 3fff2f70 00000010 3ffee1d8 00000088 3fff2fb0 00000018 3ffee1d8 3ffef928 3fff4cc0 3ffe9604 402305b8 3ffe9d3c 4022499c 3ffef3c8 4024c246 
3fff2fc0  000000b0 00000001 00000000 00000000 00000018 00000006 00000000 00000001 00000002 00000000 3ffedce8 00000000 000000b0 00000000 3fff4cc0 402328cd 
3fff3000  00000000 00000001 3ffe88dc 00000000 3ffefb28 3ffe8e58 3ffee5cc 4022ad6c 004c4b40 3fff4cc4 3ffe87d0 402103c1 00000002 0000000c 40215ed0 00000000 
3fff3040  00000001 3ffef928 3fff4cc4 402103f1 00000001 3ffef928 3fff4cc4 40231292 423a7d80 40230402 3ffe91bc 3ffef928 3ffef928 00000005 3ffee1d8 40233c62 
3fff3080  00000001 80af1999 00000007 4021ba94 00000000 00000000 3ffefa24 4010060d 00000000 00000017 60000200 4021bb31 4023368c 00000001 00000005 3ffef928 
3fff30c0  3ffee1d8 40233b4c 3ffee148 402337b9 3ffea954 40233b4c 00000000 4023382c 3ffea954 40233b4c 3ffee1d8 40233e34 3fff4cc0 40233f34 3fff4cc0 40233f34 
3fff3100  40215eec 00000002 0000000a 3ffee530 00000001 3ffee1d8 3ffee1d0 402340ce 402135cd 3ffe91bc 3fff4cc0 40233f34 00000001 a5a5a5a5 a5a5a5a5 3ffefbac 
3fff3140  402148b1 402148b4 3ffef9ac 4024c246 3ffedce8 3fff4cc0 3ffedd7b 00000001 00000000 00000000 3ffea3b8 4022ad6c 40233f34 3fff4cc0 00000000 3ffefbac 
3fff3180  40233f34 3fff4cc0 00000000 402316d0 4010738c 4010053a 00000030 40210208 3fff31e0 00000020 00000000 4022ad6c 0007a120 3ffea3b8 3ffe87d0 402103c1 
3fff31c0  4010738c 4010053a 00000030 4022ad6c 3fff3210 00000020 00000000 0007a0e3 00079be5 00000000 3ffe921c 00000003 00000009 fffff000 00000002 3ffe87d0 
3fff3200  3ffe8fe8 402319bc 3ffee148 00000013 00000000 3ffe8fe8 3ffe87d0 4022ad6c 4010738c 402104d0 00000030 402104b4 00000000 00000000 3ffe87d0 00000000 
3fff3240  00000000 00000000 3ffe87d0 402104f9 3ffe8ff8 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 a5a5a5a5 

      PC: 0x00000000        PS: 0x00000031        A0: 0x00000000        A1: 0x3ffe7fc0  
      A2: 0x00000000        A3: 0x00000000        A4: 0x3ffe7fc0        A5: 0x00000000  
      A6: 0x3ffe8f9f        A7: 0x00000007        A8: 0x3ffe8f9e        A9: 0x000000b0  
     A10: 0x3ffe8f9f       A11: 0x0000006d       A12: 0x00000000       A13: 0x00000000  
     A14: 0x00000000       A15: 0x00000000       SAR: 0x0000001f  EXCCAUSE: 0x00000014  

```

xiongyumail, 2018-08-01:
Hi @ichenxing ,
任务堆栈深度设置了多大？

ichenxing, 2018-08-01:
@xiongyumail xTaskCreate(&main_task, "main_task", 2048, NULL, 5, NULL);	// 4096 3140

ichenxing, 2018-08-01:
同一个可执行程序在不同的开发板上呈现的结果不一样，烧录配置都是一样的：
1.开发板：esp8266 S1；FLASH：2MB
```
 ets Jan  8 2013,rst cause:1, boot mode:(3,0)

load 0x40100000, len 4344, room 16 
tail 8
chksum 0xcb
load 0x3ffe8008, len 3420, room 0 
tail 12
chksum 0xf3
csum 0xf3
[0;32mI (27) boot: ESP-IDF v2.0.0-733-g3c806b7 2nd stage bootloader[0m
[0;32mI (27) boot: compile time 16:03:18[0m
[0;32mI (28) boot: SPI Speed      : 40MHz[0m
[0;32mI (31) boot: SPI Mode       : DOUT[0m
[0;32mI (35) boot: SPI Flash Size : 2MB[0m
[0;32mI (39) boot: Partition Table:[0m
[0;32mI (43) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (50) boot:  0 nvs              WiFi data        01 02 00009000 00006000[0m
[0;32mI (57) boot:  1 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (65) boot:  2 factory          factory app      00 00 00010000 00100000[0m
[0;32mI (72) boot: End of partition table[0m
[0;32mI (76) esp_image: segment 0: paddr=0x00010010 vaddr=0x40210010 size=0x4cb18 (314136) map[0m
[0;32mI (189) esp_image: segment 1: paddr=0x0005cb30 vaddr=0x3ffe8000 size=0x00ec8 (  3784) load[0m
[0;32mI (190) esp_image: segment 2: paddr=0x0005da00 vaddr=0x3ffe8ec8 size=0x0011c (   284) load[0m
[0;32mI (196) esp_image: segment 3: paddr=0x0005db24 vaddr=0x40100000 size=0x07d34 ( 32052) load[0m
[0;32mI (215) boot: Loaded app from partition at offset 0x10000[0m
[0;32mI (0) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE[0m
[0;32mI (0) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE[0m

 ============= spi init slave =============

[0;32mI (0) main_task: Setting WiFi configuration SSID SpeechX_2.4G...[0m
mode : sta(80:7d:3a:42:02:04)
add if0
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 31
pm open phy_2,type:2 0 0
cnt 

connected with SpeechX_2.4G, channel 11
[0;32mI (5) event: sta ip: 192.168.1.117, mask: 255.255.255.0, gw: 192.168.1.1[0m
[0;32mI (5) file_download: Connected to AP[0m
[0;32mI (5) file_download: DNS lookup succeeded. IP=120.79.163.252[0m
[0;32mI (5) file_download: ... allocated socket[0m
[0;32mI (5) file_download: ... connected[0m

[0;32mI (5) main_task: 1: hello world![0m
[0;32mI (6) main_task: 2: hello world![0m
[0;32mI (7) main_task: 3: hello world![0m

Core 0 was running in ISR context:

Task stack [IDLE] stack from [0x3fff0d18] to [0x3fff1014], total [768] size

          0        1        2        3        4        5        6        7        8        9        a        b        c        d        e        f        

3fff0d18  a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 
3fff0d58  a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 
3fff0d98  a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 
3fff0dd8  a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 
3fff0e18  a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 
3fff0e58  a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 
3fff0e98  a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 3ffe9ed0 00000065 00000004 00000000 401004f3 a5a5a5a5 a5a5a5a5 a5a5a5a5 40106f42 40107308 000c3500 
3fff0ed8  285ac417 a5a5a5a5 40107330 4010055e 00000030 40100631 3fff0f30 00000020 00000000 01060065 01000000 00060000 01060065 00000000 4021bb5a 3feffe00 
[... 126 lines cut ...]
[0;32mI (12) main_task: 8: hello world![0m
[0;32mI (13) main_task: 9: hello world![0m
[0;32mI (14) main_task: 10: hello world![0m
[0;32mI (15) main_task: 11: hello world![0m
[0;32mI (16) main_task: 12: hello world![0m
[0;32mI (17) main_task: 13: hello world![0m
[0;32mI (18) main_task: 14: hello world![0m
[0;32mI (19) main_task: 15: hello world![0m
[0;32mI (20) main_task: 16: hello world![0m
[0;32mI (21) main_task: 17: hello world![0m
[0;32mI (22) main_task: 18: hello world![0m
[0;32mI (23) main_task: 19: hello world![0m
[0;32mI (24) file_download: Connected to AP[0m
[0;32mI (24) file_download: DNS lookup succeeded. IP=120.79.163.252[0m
[0;32mI (24) file_download: ... allocated socket[0m
[0;32mI (25) file_download: ... connected[0m
[0;32mI (25) file_download:  INFO is: HTTP/1.1 200 

Content-Type: application/json;charset=UTF-8

Transfer-Encoding: chunked

Date: Wed, 01 Aug 2018 08:05:30 GMT

```

ichenxing, 2018-08-01:
@donghengqaz 2048不是4的倍数？

ichenxing, 2018-08-01:
可能存在的原因：使用了esp8266 S1模组作为HSPI的从机，当SPI主机来访问esp8266 S1模组（读hspi状态）时，会产生中断（调用函数spi_slave_isr_sta）才可能导致上述的任务堆栈问题？

ichenxing, 2018-08-03:
在一个例程中看见这么一句话
```
/* Temporarily pin task to core, due to FPU uncertainty */
 xTaskCreate(&aws_iot_task, "aws_iot_task", 9216, NULL, 5, NULL);
```
“ Temporarily pin task to core, due to FPU uncertainty”该如何理解？

ichenxing, 2018-08-03:
@donghengqaz @xiongyumail 我觉得不应该是任务堆栈大小的问题，我尝试过使用更大的任务堆栈，比如：8192和16384等，还是会出现上述问题；问题可能出现在esp8266的HSPI slave 中断上面，“Core 0 was running in ISR context...”是在外部SPI主机设备来访问HSPI状态寄存器的值产生的？

kmalloc8, 2018-08-03:
我使用`红外中断`也出现类似情况，加大堆栈，只会降低crash概率。。。

kmalloc8, 2018-08-03:
话说，SDK v3.0会修到什么时候才会稳定些?

27-301, 2018-08-10:
3.0 只会提供 GPIO & UART 驱动。3.1 中会提供其他驱动。

27-301, 2018-08-10:
本月会提供 3.0 release，3.1 应该在 10 月底。
不过驱动是独立的，会逐步提供出来，时间大概是 9 月份。你需要哪几个？

ichenxing, 2018-08-10:
@27-301 需要使用SPI来传输文件，要等那么久啊啊啊......

closed 2019-04-17 by FayeY