espressif/ESP8266_RTOS_SDK #273: 自搭mosquitto带ssl的服务器可以连上，但亚马逊aws iot平台无法连上。可能是什么问题？

9crk, 2018-08-02:
自搭mosquitto带ssl的服务器可以连上，但亚马逊aws iot平台无法连上。可能是什么问题？
两个都用MQTT.fx 软件可以连上。

9crk, 2018-08-02:
代码一丝没改，就改了服务器地址和ca cert key三个头文件。

9crk, 2018-08-02:
亚马逊的堵在SSL_Connect函数，返回0

9crk, 2018-08-02:
@27-301  2.0.0 Release

9crk, 2018-08-02:
@xcguang  
NetworkConnectSSL(&network, address, MQTT_PORT, &ssl_cck, SSLv23_client_method(), SSL_VERIFY_PEER, 8192);

这个8192应该就是你说的buffer吧？

另外，SSLv23_client_method()这个换过了多个，也不行。

xcguang, 2018-08-02:
如果有条件的话，建议使用无线抓包网卡抓取一下，如果是 ssl 层的错误，应该能看出来点什么的

kmalloc8, 2018-08-02:
mark, I want the answer too.

kmalloc8, 2018-08-10:
@9crk ,可以连上了吗？

27-301, 2018-08-10:
建议用 3.0，直接拉 master，有如下例子：examples/protocol/aws_iot。

kmalloc8, 2018-08-10:
@27-301 ，前些天用3.0才写的一些外设驱动，比如Key，今天就更新gpio了，以前的接口也没有了，后面会不会哪天又大改一次?

xiongyumail, 2018-08-10:
@kmalloc8 
3.0版本外设驱动大部分兼容 ESP32 外设驱动，而且接口简单了很多．可以尝试一下example中的例程．

kmalloc8, 2018-08-10:
@xiongyumail ，OK，我研究一下，thx.

kmalloc8, 2018-08-10:
@9crk ,呵，因为一定要搞aws，感觉3.0移植了官方aws-sdk，应该靠谱些吧，不管那么多了，现在开始折腾3.0，，，

closed 2018-09-03 by FayeY