espressif/ESP8266_RTOS_SDK #276: smartconfig demo 不能配网成功

zhouwei890610, 2018-08-03:
[image]

zhouwei890610, 2018-08-03:
APP 是ISO 版本Esptouch

zhouwei890610, 2018-08-03:
[image]

xuhongv, 2018-08-04:
您可以尝试下 android端的。或者保证您的 ios 包是最新的版本。

jerrywoodz, 2018-08-09:
Android Esptool v0.3.6.2是最新板？，smartconfig也配網不成功。

27-301, 2018-08-16:
Is this dual band router?

qianxilin, 2018-09-27:
我也尝试了3.0的SDK。smartconfig无法配网成功。。同样的环境2.0的SDK是正常的。。有谁使用3.0的SDK配网成功过？

Junhao-Espressif, 2018-10-09:
Hi @qianxilin,

Please use the new apk here: https://github.com/EspressifApp/EsptouchForAndroid/tree/master/releases/apk .

And select Broadcast in EspTouch app.

zhanghuichina, 2018-10-09:
有些路由器可以，有些很难成功，都没有5g的路由器

IanXxu, 2018-10-11:
用EspTouch app，smartconfig偶尔会成功，成功率很低啊。airkiss也是偶尔成功

zhanghuichina, 2018-10-11:
同样的网络环境 使用esp32时esptouch和airkiss都能成功,而且速度很快, 应该是libsmartconfig.a的问题, 有没有其他版本可以提供v3.0测试的?

yefunbox, 2018-10-14:
v3.0确实用不了，很奇怪

qianxilin, 2018-10-22:
重新用Ubuntu把环境搭建了下。EspTouch ap可以连接成功了。。把linux的SDK重新拷贝到windows下，再编译，也可以连接了。-_-||

closed 2018-10-22 by FayeY

1XiaSheng, 2018-10-24:
我在安卓端也碰到了配置不成功.
I have the same err in esptouch on android

FayeY, 2018-10-24:
使用的是最新的安卓 apk 和固件吗？可以参考上面的帖子更新。