espressif/ESP8266_RTOS_SDK #278: smartconfig crashed,after restart serveral time

aeris-ming, 2018-08-05:
Log:
```
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 2592, room 16 
tail 0
chksum 0xf3
load 0x3ffe8000, len 764, room 8 
tail 4
chksum 0x92
load 0x3ffe82fc, len 676, room 4 
tail 0
chksum 0x22
csum 0x22

2nd boot version : 1.7(5d6f877)
SPI Speed : 40MHz
SPI Mode : DIO
SPI Flash Size & Map: 32Mbit(512KB+512KB)
jump to run user1 @ 1000

OS SDK ver: 2.0.0(e271380) compiled @ Mar 30 2018 18:54:06
phy ver: 1055_1, pp ver: 10.7
rf cal sector: 1019
tcpip_task_hdl : 40107a00, prio:10,stack:512
idle_task_hdl : 40107ab0,prio:0, stack:384
tim_task_hdl : 40107bf8, prio:2,stack:512
SDK version:2.0.0(e271380)
mode : softAP(86:f3:eb:b4:6e:96)
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
add if1
bcn 100
lpush 1 3000 cleared
bcn 0
del if1
usl
sul 0 0
mode : sta(84:f3:eb:b4:6e:96)
add if0
SC version: V2.5.4
scandone
scandone
SC_STATUS_FIND_CHANNEL
lpush 1 3000 cleared
force restart
sc callback interrupted
SC version: V2.5.4
scandone
scandone
SC_STATUS_FIND_CHANNEL
lpush 1 3000 cleared
force restart
sc callback interrupted
SC version: V2.5.4
scandone
lpush 1 3000 cleared
force restart
sc callback interrupted
SC version: V2.5.4
scandone
err,scan status 1
scandone
err,scan status 4
scandone
err,scan status 1
scandone
err,scan status 4
scandone
err,scan status 1
scandone
err,scan status 4
scandone
err,scan status 1
scandone
err,scan status 4
"smartconfig_tas"(stack_size = 0,task handle = 40107e28) overflow the heap_size.
Fatal exception (20): 
epc1=0x00000000
epc2=0x00000000
epc3=0x402200f0
epcvaddr=0x00000000
depc=0x00000000
rtn_add=0x00000000

Free Heap Size: 54368
Stack Point: 3fff7010
3fff7010: 4010223b 00000000 3fff1417 401022f4
3fff7020: 00000061 3fff0dd0 3fff13c8 401003f8
3fff7030: 402278a0 3fff7090 3fff13c8 401001cc
3fff7040: 00000040 3fff13d8 3fff7090 3fff7090
3fff7050: 3fff7090 3fff13d8 00000051 40102515
3fff7060: 3fff7090 3fff7090 00000004 00000041
3fff7070: 402020c6 3ffedfb0 3ffef120 00000001
3fff7080: 00000080 00000000 00000004 4020c4bc
B%dP5鉓\'我u+S	酔蔄
\	)5[
麼RA授粇IRK呏e@蔜P)NPA	I蚟Z{僋*9∩%鈥Q=@蚘
鳱PA	I蚔P{C爷*9j藂€T撘QjU	-)
5)i(DA翾
!mN8LC壆
5)刏q衠[篮藦€5n磊!1pOS SDK ver: 2.0.0(e271380) compiled @ Mar 30 2018 18:54:06
phy ver: 1055_1, pp ver: 10.7

rf cal sector: 1019
tcpip_task_hdl : 40107a00, prio:10,stack:512
idle_task_hdl : 40107ab0,prio:0, stack:384
tim_task_hdl : 40107bf8, prio:2,stack:512
SDK version:2.0.0(e271380)
mode : softAP(86:f3:eb:b4:6e:96)
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
add if1
bcn 100
click 1
```

Source Code:
[smartconfig.zip](https://github.com/espressif/ESP8266_RTOS_SDK/files/2260414/smartconfig.zip)

xuhongv, 2018-08-06:
xTaskCreate(smartconfig_task, "smartconfig_task", 256, NULL, 2, NULL); the "256"to be replaced to 512, or more .try..

aeris-ming, 2018-08-06:
@xuhongv It seems stop_smarconfig(); not free memory

Junhao-Espressif, 2018-08-27:
Hi @aeris-ming 

Could you do not use ENTER_CRITICAL and EXIT_CRITICAL in start_smart_config task?

If still crash. Please send to me your whole project.

closed 2018-10-09 by Junhao-Espressif