espressif/ESP8266_RTOS_SDK #280: 这个问题正常吗？WARNING: Toolchain version is not supported: 1.22.0-88-gde0bdc1 Expected to see version: 1.22.0-80-g6c4433a

wikipad, 2018-08-08:
WARNING: Toolchain version is not supported: 1.22.0-88-gde0bdc1
Expected to see version: 1.22.0-80-g6c4433a
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
WARNING: Compiler version is not supported: 
Expected to see version(s): 5.2.0
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
WARNING: Toolchain version is not supported: 1.22.0-88-gde0bdc1
Expected to see version: 1.22.0-80-g6c4433a
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
WARNING: Compiler version is not supported: 
Expected to see version(s): 5.2.0
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own

27-301, 2018-08-09:
pull 一下 master，在 commit 3ffcb121 修正了这个 warning。

xuhongv, 2018-08-09:
警告而已，不必担心。

closed 2018-08-24 by 27-301