espressif/ESP8266_RTOS_SDK #284: `gpio_isr_handler_add` causes a problem ...

kmalloc8, 2018-08-13:
## Environment

- Development Kit:
- IDF version): [master]
- Development Env:      [|Make]
- Operating System:     [|Ubuntu]
- Power Supply:         [USB]

## Problem Description

In my project, I have two functions, `key_init(`) and `led_init()`, which referred to `examples/peripherals/gpio`.  If invoke `led_init()` first, and then `key_init()`, the key interrupt worked. But if invoke like `key_init();led_init();`, key interrupt didn't work.  I found `gpio_isr_handler_add` has an impact on led_init().  `led_init()` or other gpio init need to be inited before invoking `gpio_isr_handler_add`.

What's going on ? Somebody know?  
 Thanks.

xiongyumail, 2018-08-13:
Hi @kmalloc8 ,
Can you provide the implementation of key_init() and led_init()? We are not quite sure about their specific role.

kmalloc8, 2018-08-14:
Hi @xiongyumail ,  
For experiment, I used the example `ESP8266_RTOS_SDK/examples/peripherals/gpio` and just wrapped two functions.  
`void key_init(void)
{
    gpio_config_t io_conf;

    //interrupt of rising edge
    io_conf.intr_type = GPIO_INTR_ANYEDGE;
    //bit mask of the pins, use GPIO4/5 here
    io_conf.pin_bit_mask = GPIO_INPUT_PIN_SEL;
    //set as input mode
    io_conf.mode = GPIO_MODE_INPUT;
    //enable pull-up mode
    io_conf.pull_up_en = 1;
    gpio_config(&io_conf);
	
    //change gpio intrrupt type for one pin
    gpio_set_intr_type(GPIO_INPUT_IO_0, GPIO_INTR_ANYEDGE);
    //create a queue to handle gpio event from isr
    gpio_evt_queue = xQueueCreate(10, sizeof(uint32_t));
    //start gpio task
    xTaskCreate(gpio_task_example, "gpio_task_example", 2048, NULL, 10, NULL);

    //install gpio isr service
    gpio_install_isr_service(0);
    //hook isr handler for specific gpio pin
    gpio_isr_handler_add(GPIO_INPUT_IO_0, gpio_isr_handler, (void *) GPIO_INPUT_IO_0);
    //hook isr handler for specific gpio pin
    gpio_isr_handler_add(GPIO_INPUT_IO_1, gpio_isr_handler, (void *) GPIO_INPUT_IO_1);
	
}`    
  
`void led_init(void)
{
	gpio_config_t io_conf;

	//disable interrupt
    io_conf.intr_type = GPIO_INTR_DISABLE;
    //set as output mode
    io_conf.mode = GPIO_MODE_OUTPUT;
    //bit mask of the pins that you want to set,e.g.GPIO18/19
    io_conf.pin_bit_mask = GPIO_OUTPUT_PIN_SEL;
    //disable pull-down mode
    io_conf.pull_down_en = 0;
    //disable pull-up mode
    io_conf.pull_up_en = 0;
    //configure GPIO with the given settings
    gpio_config(&io_conf);

}
`  
`void app_main(void)
{

	key_init();
	led_init();
	

    int cnt = 0;

    while (1) {
        ESP_LOGI(TAG, "cnt: %d\n", cnt++);
        vTaskDelay(1000 / portTICK_RATE_MS);
        gpio_set_level(GPIO_OUTPUT_IO_0, cnt % 2);
        gpio_set_level(GPIO_OUTPUT_IO_1, cnt % 2);
    }
}`

kmalloc8, 2018-08-15:
hi @xiongyumail ,
Have you fixed the GPIO ISR issue?

xiongyumail, 2018-08-15:
Yes, in merge.I'll tell you after it's finished.

closed 2018-08-17 by wujiangang