espressif/ESP8266_RTOS_SDK #285: 关于MQTT问题，会导致断开wifi连接..

xuhongv, 2018-08-14:
如果设备已经连接服务器，但是长时间没有发布消息的话，会发生与路由器断开连接，串口打印如下：

```
bcn_timout,ap_probe_send_start
ap_probe_send_over, rest wifi status to disassoc
state: 5 -> 0 (1)
rm 0
pm close 7 0 0/626761011
reconnect
scandone
no aliyunTextOnlyNone found, reconnect after 1s
reconnect
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
```
- 请问是什么问题呢

ustccw, 2018-08-14:
@xuhongv 这个问题已找到原因，会尽快修复。

xuhongv, 2018-08-14:
@ustccw 好的。我使用的是 v2.0的分支，尽快修复下吧。这边项目蛮急的..

xuhongv, 2018-08-18:
@ustccw 请问这个问题还有得救么？
```

```

xuhongv, 2018-08-18:
```
bcn_timout,ap_probe_send_start
ap_probe_send_over, rest wifi status to disassoc
state: 5 -> 0 (1)
rm 0
pm close 7 0 0/626761011
reconnect
scandone
no aliyunTextOnlyNone found, reconnect after 1s
reconnect
scandone
```

xuhongv, 2018-08-18:
如果你们有依然不修改静态库文件的解决方法，不胜感激涕零请教。要不然我要拿 nonos的跑了....

27-301, 2018-08-21:
@xuhongv 用附件试试。
[libphy.zip](https://github.com/espressif/ESP8266_RTOS_SDK/files/2305573/libphy.zip)

xuhongv, 2018-08-21:
非常感谢。我现在试试

ustccw, 2018-08-21:
@27-301 @xuhongv 抱歉，未及时跟进此 issue.

ustccw, 2018-08-21:
@xuhongv 我将继续跟进此问题

ustccw, 2018-08-27:
@xuhongv 1055_7 版本的 libphy.a 长时间运行会存在 crash 情况，请使用下面 1055_8 版本 libphy.a,
log 开始会打印 `phy ver: 1055_8` 
[libphy.a.zip](https://github.com/espressif/ESP8266_RTOS_SDK/files/2322162/libphy.a.zip)

xuhongv, 2018-08-27:
好的。我觉得这个rtos的长时间运行会断开与MQTT服务器的连接，比 NONOS的稳定性差很多，请问你们有考虑到重连的机制么？  @ustccw

ustccw, 2018-08-27:
@xuhongv 
1. 重连机制，最好根据客户的逻辑自己添加，否则 demo 显得就是一个解决方案。
2. 更新到新的 libphy.a，理论上稳定性会好很多，你可以试下，看是否仍然存在问题。

xuhongv, 2018-08-27:
明白。我现在做了断电重连的代码，关于MQTT的代码如下：
```
static void Task_MqttClient_Connect(void *pvParameters) {

	Network network;
	unsigned char sendbuf[2048], readbuf[2048] = { 0 };
	int rc = 0, count = 0;
	MQTTPacket_connectData connectData = MQTTPacket_connectData_initializer;
	pvParameters = 0;
	NetworkInit(&network);
	MQTTClientInit(&client, &network, 5000, sendbuf, sizeof(sendbuf), readbuf,
			sizeof(readbuf));

	bool first_flag = false;

	char *address = MQTT_SERVICE;
	connectData.MQTTVersion = 3;
	connectData.clientID.cstring = originTopic;
	connectData.username.cstring = MQTT_USER_NAME;
	connectData.password.cstring = MQTT_USER_PAW;
	connectData.keepAliveInterval = 30;
	connectData.cleansession = true;

	for (;;) {

		while (wifi_station_get_connect_status() != STATION_GOT_IP) {
			vTaskDelay(1000 / portTICK_RATE_MS);
		}

		if (IS_DEBUG_OPEN)
			printf("Task_MqttClient_Connect NetworkConnect init...\n");

		if ((rc = NetworkConnect(&network, address, MQTT_PORT)) != 0) {
			if (IS_DEBUG_OPEN)
				printf("MClouds NetworkConnect connect fail , code %d\n", rc);
		} else {
			if (IS_DEBUG_OPEN)
				printf(" NetworkConnect init succeed...\n");
		}

		if (!first_flag)
			if ((rc = MQTTStartTask(&client)) != pdPASS) {
				if (IS_DEBUG_OPEN)
					printf("Return code from start tasks fail , errorCpode: %d\n",
							rc);
			} else {
				if (IS_DEBUG_OPEN)
					printf("Use MQTTStartTask succeed...\n");
				first_flag = true;
			}

		if ((rc = MQTTConnect(&client, &connectData)) != 0) {
			if (IS_DEBUG_OPEN)
				printf("[SY] MClouds connect fail , errorCode: %d\n", rc);
			network.disconnect(&network);
			vTaskDelay(1000 / portTICK_RATE_MS);
		}

		if ((rc = MQTTSubscribe(&client, subTopic, QOS0, MessageArrived))
				!= 0) {
			if (IS_DEBUG_OPEN)
				printf("[SY] MClouds Subscribe fail is %d\n", rc);
			network.disconnect(&network);
			vTaskDelay(1000 / portTICK_RATE_MS);
		} else {
			isConnectMQTT = true;
			if (IS_DEBUG_OPEN)
				printf("MQTT subscribe to topic -> %s\n", subTopic);

		}

		xQueueReset(MqttMessageQueueHandler);

		while (1) {

			char payload[2048];
			struct esp_mqtt_msg_type *pMsg;
			printf("MqttMessageQueueHandler waitting ..\n");

			xQueueReceive(MqttMessageQueueHandler, &pMsg, portMAX_DELAY);

			if (pMsg->power == 3) {
				rc = 1; //标志位置位1，准备重连
			} else {
				sprintf(payload, "%s", pMsg->allData);
				MQTTMessage message;
				message.qos = QOS1;
				message.retained = false;
				message.payload = (void *) payload;
				message.payloadlen = strlen(payload) + 1;

				if ((rc = MQTTPublish(&client, pubTopic, &message)) != 0) {
						printf("MQTT publish succeed fail %d\n", rc);
				} else {
						printf("MQTT publish succeed  ...\n");
				}
			}
			if (rc != 0) {
				break;
			}
			printf("network: %d\n", rc);
		}
		if (IS_DEBUG_OPEN)
			printf("network.disconnect\n");
		network.disconnect(&network);
		vTaskDelay(1500 / portTICK_RATE_MS);
	}
	
	printf("mqtt_client_thread going to be deleted\n");
	mHandlerMQTTConntion = NULL;
	vTaskDelete(NULL);
	return;
}
```

然后我是用硬件定时器做了每 15 秒轮询是否断开MQTT服务器，如果是则发送消息去让上面自己去重新连接，存在如下问题：
- [x] 上述代码是先断开底层的````TCP```服务连接，然后重新连接服务器；成功率还好，五次测试有三次是可以重新连接成功的效果；
- [x] 正常是执行三次或四次重连成功后，下一次就会重连不成功，打印就到了 ``` NetworkConnect init succeed...```，之后就应该卡住了，之后打印如下信息，稍后系统就重启了。应该是静态库文件发送问题了，还望检查下。
``` ShowCritical:0
HdlMacSig:1
  SWReq:1

```
-----------------------------------
- 执行三到四次之后，提示：

```
 now send reConnect msg 
 network.disconnect
 Task_MqttClient_Connect NetworkConnect init...
 NetworkConnect init succeed...

 [SY] now time: 1 , 16:5:0 , get freeHeap: 39768 , client.isconnected: 0
now send reConnect msg 
 
 [SY] now time: 1 , 16:5:15 , get freeHeap: 39768 , client.isconnected: 0
now send reConnect msg 

ShowCritical:0
HdlMacSig:1
SWReq:1

B-l*酠%掏	%[)S	酔蔆	)n*)NRA	I蚔嘀TZ@蔜@蚘
鳯奀	KRK吷F	)l€NRA	I蚔T愙VTp-蒠5誐Yp@H7謪)i@@華蔘
)eD€@華婣
AMi
)刏q襮@蒚PV5f汹!	pOS SDK ver: 2.0.0(cec1a75) compiled @ Aug 20 2018 10:31:05

```

------------------------------------------------------------------

- [image]

ustccw, 2018-08-27:
@xuhongv 这个 issue 开始是解决 beacon timeout 问题的，如果没有再出现 beacon timeout 的问题。可以关闭这个 issue。

剩下的断线重连是另外的问题，麻烦重新 new 一个 issue 来跟进问题，谢谢。

xuhongv, 2018-08-27:
好的。更新了这个库文件之后，之前那个问题的确关闭了，非常感谢您！！
@ustccw 
@27-301

closed 2018-08-27 by xuhongv