espressif/ESP8266_RTOS_SDK #286: Random crashes and assertions in LWIP

THKDev, 2018-08-15:
## Environment

- Development Kit: ESP12-E
- IDF version (``git rev-parse --short HEAD`` to get the commit id.): fd63b5e561d76415d70fd9d1dea0a744e86ea66a
- Development Env:      Make
- Operating System:     Fedora
- Power Supply:         external 3.3V

## Problem Description
random crashes in tcp_write or tcp_out . Log file lines 22195, 34847, 35207 and 40776. Or search for a5a5a5 .

### Expected Behavior
no crashed would be fine.

### Actual Behavior
random crashes.

## Debug Logs
[log.zip](https://github.com/espressif/ESP8266_RTOS_SDK/files/2290798/log.zip)

```
      PC: 0x40246f61        PS: 0x00000030        A0: 0x40246f5d        A1: 0x3fff1750  
0x40246f61: tcp_output at /home/user/projects/esp/ESP8266_RTOS_SDK/components/lwip/lwip/src/core/tcp_out.c:1154
0x40246f5d: tcp_output at /home/user/projects/esp/ESP8266_RTOS_SDK/components/lwip/lwip/src/core/tcp_out.c:1154

      A2: 0x00000000        A3: 0x00001f00        A4: 0x00001f00        A5: 0x00000000  
      A6: 0x3ffea104        A7: 0x00000000        A8: 0x00000003        A9: 0x00000000  
     A10: 0x00000000       A11: 0x00000000       A12: 0x3fff0980       A13: 0x3fff0858  
     A14: 0x00001fae       A15: 0x00000067       SAR: 0x0000001d  EXCCAUSE: 0x0000001c  ```

## Other items if possible

- [x] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

donghengqaz, 2018-08-16:
Could you apply your “elf” to us? And then we can track the problem easily.

27-301, 2018-08-16:
It will be better if you can provide the demo code.

THKDev, 2018-08-19:
New log file and small test program. Sometimes the program crashs before running `app_main(`) or later while doing dns lookup by `dns_gethostbyname` .
Maybe some restarts are necessary to trigger a crash. I would say 1 of 5 restart will crash. See log file.

[log.zip](https://github.com/espressif/ESP8266_RTOS_SDK/files/2300311/log.zip)
[sample.zip](https://github.com/espressif/ESP8266_RTOS_SDK/files/2300312/sample.zip)

THKDev, 2018-08-23:
Another test with [master](https://github.com/espressif/ESP8266_RTOS_SDK/tree/bffb2b166229edd7f5c99a236d013d48bfcd1e7d)
After one successful boot; next boot will hang in chip_init . Unfortunately code of method chip_init is not open source.
[serial2.zip](https://github.com/espressif/ESP8266_RTOS_SDK/files/2315672/serial2.zip)

THKDev, 2018-08-26:
Some more crash logs from task with name [ppT] and [tiT] .
[crashlogs.zip](https://github.com/espressif/ESP8266_RTOS_SDK/files/2321411/crashlogs.zip)

FayeY, 2018-10-22:
Hi, could you have a try with our latest SDK？It should work fine, does not have this issue.

THKDev, 2018-10-24:
Looks good at the moment. Thanks for your work!

closed 2018-10-24 by THKDev