espressif/ESP8266_RTOS_SDK #288: I'm not able to work with this toolchain

yash0daniel, 2018-08-18:
----------------------------- Delete below -----------------------------

If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at bbs.espressif.com instead.

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [ ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ ] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs on the monitor](http://esp-idf.readthedocs.io/en/latest/get-started/idf-monitor.html#automatically-decoding-addresses).
4. Provide more items under **Other items if possible** can help us better locate your problem.
5. Use markup (buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:
- IDF version (``git rev-parse --short HEAD`` to get the commit id.): 
    //bd6ea4393c7d2f059fc4decc70f1ec3eb3597268
- Development Env:      [Arduino IDE|Make|Eclipse|other]
- Operating System:     [Windows|Ubuntu|MacOS]
- Power Supply:         [USB|external 5V|external 3.3V|Battery]

## Problem Description

//Detailed problem description goes here.

### Expected Behavior

### Actual Behavior

### Steps to repropduce

1. step1
2. ...

// It helps if you attach a picture of your setup/wiring here.

### Code to reproduce this issue

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
#include "esp_log.h"

void app_main()
{
    
}

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

```
Debug log goes here.
Please copy the plain text here for us to search the error log. Or attach the complete logs but leave the main part here if the log is *too* long.
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

yash0daniel, 2018-08-18:
yash@daniel:~/esp/ESP8266_RTOS_SDK/examples/get-started/project_template$ make menuconfig
/bin/sh: 1: xtensa-lx106-elf-gcc: not found
/bin/sh: 1: xtensa-lx106-elf-gcc: not found
WARNING: Failed to find Xtensa toolchain, may need to alter PATH or set one in the configuration menu
MENUCONFIG

*** End of the configuration.
*** Execute 'make' to start the build or try 'make help'.

yash@daniel:~/esp/ESP8266_RTOS_SDK/examples/get-started/project_template$ make all
/bin/sh: 1: xtensa-lx106-elf-gcc: not found
/bin/sh: 1: xtensa-lx106-elf-gcc: not found
WARNING: Failed to find Xtensa toolchain, may need to alter PATH or set one in the configuration menu
GENCONFIG
/bin/sh: 1: xtensa-lx106-elf-gcc: not found
/bin/sh: 1: xtensa-lx106-elf-gcc: not found
WARNING: Failed to find Xtensa toolchain, may need to alter PATH or set one in the configuration menu
CC build/bootloader/main/bootloader_start.o
make[2]: xtensa-lx106-elf-gcc: Command not found
/home/yash/esp/ESP8266_RTOS_SDK/make/component_wrapper.mk:285: recipe for target 'bootloader_start.o' failed
make[2]: *** [bootloader_start.o] Error 127
/home/yash/esp/ESP8266_RTOS_SDK/make/project.mk:467: recipe for target 'component-main-build' failed
make[1]: *** [component-main-build] Error 2
/home/yash/esp/ESP8266_RTOS_SDK/components/bootloader/Makefile.projbuild:40: recipe for target '/home/yash/esp/ESP8266_RTOS_SDK/examples/get-started/project_template/build/bootloader/bootloader.bin' failed
make: *** [/home/yash/esp/ESP8266_RTOS_SDK/examples/get-started/project_template/build/bootloader/bootloader.bin] Error 2

xiongyumail, 2018-08-18:
Hi
Looks like there is no path to set the build toolchain, please read the README settings carefully.

xiongyumail, 2018-08-18:
`export PATH=&PATH:your_tool_chain_path/bin`

kmalloc8, 2018-08-18:
export PATH=path/to/your/xtensa-lx106-elf/bin:$PATH

27-301, 2018-09-14:
Close, you can reopen if the issue still exists, thanks.

closed 2018-09-14 by 27-301