espressif/ESP8266_RTOS_SDK #289: can not find sdk/tools/windows/eclipse_make.py when build in eclipse use sdk-v3.

zaazbb, 2018-08-23:


zaazbb, 2018-08-24:
and I have some more questions about SDK-V3:
1, Can I put two app partitions (ota_0 and ota_1) in one 1M flash? partitions map like below:
```
nvs,      data, nvs,     0x9000,  0x4000
otadata,  data, ota,     0xd000,  0x2000
phy_init, data, phy,     0xf000,  0x1000
ota_0,    0,    ota_0,   0x10000, 0x70000
ota_1,    0,    ota_1,   0x90000,0x70000
```
2,  I do not know how to do app upgrade?
 In SDK-V2, the `system_upgrade_start()` do the most things like download upgrade file and write flash, use don't need do any thing else. but in SDK-V2, I can't find this function. I need a app upgrade example for SDK-V3.

thank you.

27-301, 2018-09-14:
hi, @zaazbb For eclipse_make.py, you can get from esp-idf firstly, https://github.com/espressif/esp-idf/blob/master/tools/windows/eclipse_make.py.

For config eclipse, you can refer to https://docs.espressif.com/projects/esp-idf/en/latest/get-started/eclipse-setup.html

We don't support to put ota_0 and ota_1 in a single 1MB space, it's in our plan, and will be available in v3.2.

For upgrade, please refer to https://github.com/espressif/ESP8266_RTOS_SDK/tree/master/examples/system/ota

FayeY, 2018-10-23:
Hi, 1MB flash OTA is supported in the latest master, please have a try. Thanks.

closed 2018-10-23 by FayeY