espressif/ESP8266_RTOS_SDK #291: softAP模式下，持续发送udp，过一段时间（不到两个小时），wifi就连接不上

tianhuanz, 2018-08-27:
处于这种状态下，看门狗也不会自己复位，udp也不能继续发送，wifi连接列表可以看到接入点名称，但是连接不上。

27-301, 2018-08-27:
which commit do you use? can you share the sample code and test flow?

tianhuanz, 2018-08-27:
i have try the lastest commit.

tianhuanz, 2018-08-27:
i just redirect ESP_LOG to a UDP port. so i can use the uart0.

xiongyumail, 2018-08-27:
建议周期打印当前heap大小，观察heap的变化．可能内存空间没有释放．

tianhuanz, 2018-08-27:
应用代码并没有使用malloc，如果不使用UDP，能够正常运行。

donghengqaz, 2018-08-28:
你把你的工程发给我门看看，不方便发源码，发个lib也可以。

xuhongv, 2018-08-28:
估计内存出问题吧？ 频繁调用```system_get_free_heap_size()```看看吧？

tianhuanz, 2018-08-28:
[simple_wifi.zip](https://github.com/espressif/ESP8266_RTOS_SDK/files/2327875/simple_wifi.zip)

我写了个测试代码。
[udplogger.zip](https://github.com/espressif/ESP8266_RTOS_SDK/files/2327885/udplogger.zip)

tianhuanz, 2018-08-29:
[simple_wifi.zip](https://github.com/espressif/ESP8266_RTOS_SDK/files/2330283/simple_wifi.zip)
更新一下测试代码，前面那个版本被文档管理系统加密了。

Junhao-Espressif, 2018-08-29:
Hi @tianhuanz 

Sorry to tell you that I can not download your code.
```
This site can’t be reached
github-production-repository-file-5c1aeb.s3.amazonaws.com refused to connect.
```
Please send your code to my email:  zhangjunhao@espressif.com

Could you change ```#define DHCPS_LEASE_TIME_DEF (120)``` to ```#define DHCPS_LEASE_TIME_DEF (360)``` and test again?

Please give us the error log.

tianhuanz, 2018-08-30:
DHCPS_LEASE_TIME_DEF (360) 问题依然存在

tianhuanz, 2018-09-02:
@Junhao-Espressif 已经给你们发了测试代码，请问有什么进展？

Junhao-Espressif, 2018-09-10:
Hi @tianhuanz 

Please do not call sendto in udp sentto callback function.

It does not take two hours. Just use another station connect to softap, and disconnect with softap. The softap will hang.

You should refactor udpLoggerPutc function. Write data to a send buffer and send it in another task.

FayeY, 2018-09-18:
Is there any update?

closed 2018-10-09 by Junhao-Espressif